davidramiro/Marlin-Ai3M #2: com port connetcions problem

1f7, 2018-12-14:
Hi @davidramiro 
today i flash Marlin-AI3M-181213-TMC2208.hex. connectors in original orientation.  
Step by step i configure like written on https://www.thingiverse.com/thing:3249319  
exept "Bonus: Manual Mesh Bed Leveling".  
Reboot, try to connect from Octoprint with error:  
````bash
Connecting to: /dev/ttyUSB1   
Changing monitoring state from "Offline" to "Opening serial port"  
Connected to: Serial<id=0xaa976cf0, open=True>(port='/dev/ttyUSB1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor  
Changing monitoring state from "Opening serial port" to "Connecting"  
Send: N0 M110 N0*125  
Recv: �Np|�\x18_�M��hn6H5\x06\x07JpON6��Z�z-\x0c>|YL6M>~HN6a.�8\x0f\x04\x07\x1a~�H\x07�H�X�J\x13\x03�~�Z-�]]�M�7��5\x08\x07  
Recv: \x08�^�\x1b6\x17jO>\x0b��H7�(^�Z��\x07J�%\x08\x08\x04p|"]\x0bX\x1b6�H.l�p|\x084\x08]�f�J�J�X�\x1a�H\x13H6f�J�JJ�8J�(K�J?Jzp|$~H\x0f^\x17�}�$p|\x084Recv: ��I��I�k\x08�HJp|$~\x18�\x0b^>h}�zf�J�Z�A\x08�A\x08nh%�X�\x1a�\x0b^>h}�z�Kq?.aKP�..aKKa�.a+\x07J�J�%HJ�AHJ�AHJp|@_a\x13�6�\x0fM\x7f\x1fY�6]�6\x1b�6O6�]�6\x1b�&>�K7&\x1f�K{&>�K\x1b&��+\x07J�Z�J�\x18�Jj\x08N�8�Jj\x08��X�*�Ma*\x07J�J�I�j\x08N�X�:\x17\x08\x0bd_6�p|\x080\x08@�I�\x07\x1a]N6�\x0b*\x07J�K�If�  
There was a timeout while trying to connect to the printer  
Changing monitoring state from "Connecting" to "Offline"  
Connection closed, closing down monitor
````  
what is a dirty answer? maybe i missed something?

davidramiro, 2018-12-14:
Hi Vladimir,

change the baudrate to 250000 and try again please.

[image]

1f7, 2018-12-14:
Hi David,
Thank you for quick answer.
Yes, I clone repo and found 250000. Auto for some reason did not work, set manually.
Thanks for best Cubic mod!

closed 2018-12-14 by 1f7

davidramiro, 2018-12-14:
Glad you got it to work.

Cheers!