davidramiro/Marlin-Ai3M #3: Cooling fan work only with S255

1f7, 2018-12-14:
One more problem found.
Cooling fan work only with M106 S255. Any less 255 stop fan. I dont know maybe it Sunon fan problem.
Found in http://3dtoday.ru/questions/obduv-detali-rabotaet-tolko-na-maksimume/#238793 about FAN_SOFT_PWM works better.
Can you check own and if works please write fan part number. 
Or please write how configure source for build firmware with connectors in original orientation. 
Thank you!

davidramiro, 2018-12-14:
Hi Vladimir,
Thanks for the feedback. Seems like the Sunon fan just doesn't work with stock PWM mode, as you found out yourself, SOFT_PWM should work better.

The fan I use is a DELTA BFB04512HHA, it's the stock fan. With mine, I start it with 30% for the first layer (M106 S76) and it works without a problem. 

To adjust the Configuration.h for flipped connectors, see line 855-863. https://github.com/davidramiro/Marlin-AI3M/blob/master/Marlin/Configuration.h#L854

You need to flip true and false for X,Y,Z, E0 and E1:
```
#define INVERT_X_DIR true
#define INVERT_Y_DIR false
#define INVERT_Z_DIR false
#define INVERT_E0_DIR false
#define INVERT_E1_DIR false
```

1f7, 2018-12-15:
> BFB04512HHA  

I got a very noisy fan and I decided to change it. I only had sounon.

Thank you very much, I will try play with SOFT_PWM or buy normal cooler.

davidramiro, 2018-12-15:
Actually, many people use Sunon 5015 fans as a replacement for their parts cooling fan, most of them do work without SOFT_PWM afaik. Maybe it's just your particular model or unit that doesn't. 

Anyways, please let me know if you get it to work.

1f7, 2018-12-15:
Was waiting while printing cam holder.

Yes in work.
Only uncomment one line
#define FAN_SOFT_PWM
(//#define FAST_PWM_FAN already commented)
no more.
Result minimum value for my fan work is S55
I'l happy, I like this project  :+1: 
@davidramiro Thank you very much!!!

davidramiro, 2018-12-15:
Great! And thanks for your feedback.

closed 2018-12-15 by davidramiro