davidramiro/Marlin-Ai3M #4: Issues with SD init fail

barecool, 2018-12-23:
with the new addition of the folder menu 
i cannot get the sd card to read 

on terminal it shows sd init fail?

have you ran into this.

davidramiro, 2018-12-24:
Hi,

can't reproduce this issue, sorry. After quick search through Marlin's issue section, `sd init fail` seems to point more towards a hardware issue, can you try another SD card? Also, the i3 Mega's SD slot is very fragile, it might also be an issue there. Try not fully inserting the card.

To rule out a firmware issue, leave the card inserted and flash a FW you know worked before and check again.

Also, please be aware that any kind of special character in a file or folder name will cause issues.

davidramiro, 2018-12-28:
Any news? Please report back, I'd like to know if this is an issue on your end.

barecool, 2019-01-03:
Hi

Hi ran some checks and the SD card reader has failed

It’s not your firmware

I went back to stock and it also failed

Thanks for the support

You can close the issue

davidramiro, 2019-01-03:
That's unfortunate. Try contacting Anycubic, they have great aftersales support, I already used their support and got a quick replacement on another part. They even have a video on their Youtube channel on how to swap the SD reader: https://www.youtube.com/watch?v=NGXg_dwwpJ8

Hope you'll sort this out.

closed 2019-01-03 by davidramiro