davidramiro/Marlin-Ai3M #5: Printer stops while printing because 'Heating failed', connection closes

Arasthel, 2018-12-28:
Hi, I'm using the stock drivers. This firmware works really well, but I've already had 3 times when the print process was stopped because of an error. This time I actually got a log:

```
Recv:  T:192.60 /200.00 B:60.21 /60.00 @:113 B@:22
Recv: ok
Send: N359 M205 X8 Y8*42
Recv: ok
Send: N360 G1 F2400 X115.768 Y68.936 E210.31436*57
Recv: ok
Send: N361 G1 X116.214 Y69.382 E210.35842*90
Recv: ok
Recv: ok
Send: N362 G1 X180.787 Y69.382 E214.8686*101
Send: N363 G1 X181.233 Y68.936 E214.91266*81
Recv: ok
Send: N364 G1 X181.233 Y58.663 E215.63019*92
Recv: Error:Heating failed, system stopped! Heater_ID: 0
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Printing" to "Error: Heating failed, system stopped! Heater_ID: 0 - Printer halted. kill() called!"
Changing monitoring state from "Error: Heating failed, system stopped! Heater_ID: 0 - Printer halted. kill() called!" to "Offline (Error: Heating failed, system stopped! Heater_ID: 0 - Printer halted. kill() called!)"
Connection closed, closing down monitor
```

As far as I know everything was heated properly, but I wasn't really paying attention to the temperatures while printing. The logs show that the actual temperature of the hotend was a bit lower than expected, but I don't think that is related.

Please ask for any additional logs / info if you need it.

Thanks.

Arasthel, 2018-12-28:
Actually, after researching a bit more I think this is probably not related to the FW. Feel free to close the issue if that's the case.

davidramiro, 2018-12-28:
That's just a native Marlin feature - If your heater fails to reach or hold a certain temperature over a certain amount of time, it will cancel the print as a security measure.

Not really related to my particular firmware, though we can still try to work this out. Did you do a PID tune? Those might sometimes result in less stable temperatures or even failing to reach desired temperatures. Try using the default values from the readme.
Did you limit your parts cooling fan? As you know, this Firmware unlocks the full 12V, which might be too much cooling on the stock parts cooler. You should set it to 60-70%.

Arasthel, 2018-12-29:
Thanks, I re-adjusted the cooling fan speed to 70% as suggested, and restored the default values for the PID. I hope that fixes it, thanks for your help!

closed 2018-12-29 by Arasthel