davidramiro/Marlin-Ai3M #6: First lines are not printed.

schnello, 2018-12-29:
Hello.

I want use your FW but it doenst work.

I upload the FW and i try to print my cal cube.
1. Printer start to draw the skirt > All OK
2. Printer starts to draw the fist outer lines > NOK. No material is extruded. Extruder motor stands still.
3. Printer starts to draw the infill. > OK

How i flash the FW.
I flash the FW with cura (Marlin-AI3M-181227-stock_drivers.hex)
I issue the command M502
I issue M500
I enter my custom stettings M92 X79.7607 Y80.04 Z399.92 E95.10
I store this with M500

If i fall back to the orignal FW all works fine. Any idea how i can solve this issue? What am I doing wrong?

davidramiro, 2018-12-30:
Can you please send me that particular GCode so I can try it out myself?

davidramiro, 2018-12-31:
Didn't extrude as well for me. Found the issue: As I mentioned in the description, the parts cooling fan is way stronger now on the full 12V, so you'd need limit it to around 60-70% if you are using the stock cooler. The nozzle cools down too far and the console reads
```
Send: N634 G1 F1800 X113.429 Y107.753 E80.2099*13
Recv: echo: cold extrusion prevented
```

Anyways, since you are printing at a fairly low temperature already - I guess you're printing PLA - I'd recommend not going for full cooling on the first layer already, since that can cause warping. My recommendation would be something like this:

1. layer: 0% fan speed
2. layer: 30% fan speed
3. layer: 45% fan speed
4. layer: 65% fan speed

Please report back if you're giving this firmware another try. For now, I am closing this issue since the reason was found, please feel free to reopen it if the issue still persists for you on adjusted fan settings.

closed 2018-12-31 by davidramiro

schnello, 2018-12-31:
Thanks a lot. I reflashed the FW with the new Fan settings and it works now as expected. 
Thank you for the gread work and i wish you an "Frohes neues Jahr" 👍

davidramiro, 2018-12-31:
That's good to hear. Ebenso! :tada: