davidramiro/Marlin-Ai3M #7: Y coordinates out off sync after printing

bkappler69, 2018-12-31:
Hi,

thanks for your great work! This FW is much better than the stock FW! I installed the latest version (December 27th, only used the stock version before) and have one issue on my Anycubic Mega I3:

After the print completes the following G-Code is executed (Cura 3.6 default setting):

```M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M107
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops so the head is out of the way
G1 Y180 F2000
M84 ;steppers off
G90
M300 P300 S4000
```
The `G1 Y180 F2000` command is supposed to move the plate to the front for easier access to the printed object. However, with your FW the blade moves far too much, hitting the mounts and making the mechanics jump (there's no end switch on this direction).

Sending a `G0 Y100` command to the printer positions it approximately at where the `Y100` is supposed to be.

Please note, that before printing the positioning works correctly. The only change I did to the start up G Code was to include the settings for levelling:
```
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
M501 ; restore bed leveling
M420 S1
G1 Z15.0 F{speed_travel} ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F{speed_travel}
M117 Printing...
G5
```
I can provide the full G-Code if required. The print I did was: https://www.thingiverse.com/thing:2376990

Best Regards
   Bernhard

notching, 2018-12-31:
I see this issue as well.

davidramiro, 2018-12-31:
Hi Bernhard,

thanks for the feedback. Usually I don't use Cura, but as a quick test I sliced and printed a small object with your end code and Cura 3.6.0. For me, it stopped right at the edge of the bed, just where it is supposed to, without any grinding noise or similar.

First of all, please make a copy of your EEPROM by sending `M503`. Post it on here and save it somewhere. Next up, issue a `M502` (clear EEPROM) and after that, enter all of your G29 data line by line from your previously saved output and save with `M500` to retain your mesh. Reboot your printer and try again please.

Slice some small object for me (something that only takes a few minutes to print would be great for debugging) and if it happens again, send me the GCode. I will only adjust temperatures and try it out myself.

Kind regards,
David

davidramiro, 2019-01-01:
Also, feel free to try this test build: 
[190101-test](https://kore.cc/i3mega/download/testing/190101/)

 I have implemented a minor change to the bed size, not sure if this issue has something to do with this.

davidramiro, 2019-01-02:
Possibly fixed in latest commit f877f1201fad0412afec03be7c8416818fc0aa7b, please report back after flashing [v19.01.02](https://github.com/davidramiro/Marlin-AI3M/releases/tag/v19.01.02) and resetting EEPROM (`M502` + `M500`).

Cheers.

bkappler69, 2019-01-03:
I did an some additional test prints, but I was not able to reproduce the issue again. I've now upgraded to v19.01.02 and ran the leveling test (G26 C H200 P25 R25) which was fine. Maybe the issue was caused because I did not reboot after the initial levelling?

One additional question. If levelling is enabled and I run levelling a second time (to validate and adjust), will this use the information from the previous levelling for positioning to the individual positions, or will it run from scratch?

davidramiro, 2019-01-03:
>Maybe the issue was caused because I did not reboot after the initial levelling?

Possibly, I was only recently made aware that this could be the cause - since MMBL disables endstops altogether. Weird that it doesn't only do it for Z, but I have still added it to the instructions.

>One additional question. If levelling is enabled and I run levelling a second time (to validate and adjust), will this use the information from the previous levelling for positioning to the individual positions, or will it run from scratch?

No, it will do the whole 25 points again. You can print out your mesh with `G29 S0` and adjust a single point with `S3 Xn Yn Zn.nn` - only by estimation though, you can't make the printer probe that point again, sadly.

I consider this issue resolved for now, but feel free to chime in again.

davidramiro, 2019-01-21:
Reopening this issues as some more bug reports came in.

Whether this is an issue of wrong coordinates or caused by the homing command on the end GCode in conjunction with `RESTORE_LEVELING_AFTER_G28` (which messes with endstops - software and physical ones) still needs to be checked - I strongly suspect the latter. 

I created a branch to test possible fixes - unfortunately I don't use Cura and the few test prints I've done didn't show the issue yet, hence why any feedback is highly appreciated. I'll compile it in a minute.

davidramiro, 2019-01-21:
Compiled binaries for c6c4bec8e02d096b12561059ba15eb8124d4fb8b can be found here:

https://kore.cc/i3mega/download/testing/issue-7/

davidramiro, 2019-01-22:
It was indeed the coordinates getting reset. The process of restoring leveling data resets the positions on any call of G28, even if it's just to home X like in Cura's end Gcode.

Resolved in release [v19.01.22](https://github.com/davidramiro/Marlin-AI3M/releases/tag/v19.01.22).

closed 2019-01-22 by davidramiro