davidramiro/Marlin-Ai3M #9: Behaviour when stopping (by user) in mid-print

oopeekoo, 2019-01-06:
Thanks for the great FW! 
I have (in addion this new FW) new extruder and bowden tube so I have made several test prints lately and I stopped many of them in mid-print manually from the screen. The behaviour of the Z axis is a bit odd in my opinion. When manually stopped, the Z axis moves down a bit (1 mm or so) and the hot end dibs into print and stays there. I have not tried but it might be that if print is stopped just after first layers, the nozze could crash into the bed... if not saved by endstops.
I do not remember if official FW had this behaviour too. If this is by design, I wonder what is the benefit?

davidramiro, 2019-01-06:
Hi,

assuming you are talking about aborting prints and not just pausing, I can imagine what you are seeing is caused by the firmware turning off the motors. Is it actively moving down or rather sagging a bit downwards? 

You can reproduce this behaviour by moving Z up a bit and sending `M84`. This turns off the motors, you will hear the torque being released and the Z axis might sag downwards a tiny bit. If this is what you are seeing when aborting a print, I will update the stopping behaviour (e.g. raise Z by 5mm and only then turn off motors). 

If Z actively moves down after stopping, I might have to investigate further - I don't use the TFT screen at all so I didn't run into this issue myself. Thanks for bringing it to my attention!

oopeekoo, 2019-01-07:
Did the testing. Moving Z up and then sending M84 disengaged the motors but the movement of the nozzle was very small. Then I started a test print and after few mm:s stopped (aborted) it from the touch screen. The nozzle did a dip into the print but the movement is actually so small that it could just be the motors disengaging. However, the heat of the nozzle makes noticeable damage to the print surface which, of course, is not a big deal for aborted print. But you have to move Z up before you can safely home the axis. It's not a big deal but in my opininion it could make sense to change the stopping behaviour a bit.

davidramiro, 2019-01-07:
>It's not a big deal but in my opininion it could make sense to change the stopping behaviour a bit.

Absolutely. Done :blush:
Feel free to try out the newest [release](https://github.com/davidramiro/Marlin-AI3M/releases/tag/v19.01.07).

oopeekoo, 2019-01-10:
Sorry for late responce, been busy printing not testing. The new FW does move the hot end after stopping but in my 3 tests the movement was in X axis twice, once right and once left, and once on Y axis, back. Not once movement up in Z. So the movement appears to be in random direction (or dircection of movement when stopping...). This happens in my case anyway. Has anyone else tested?

davidramiro, 2019-01-10:
Whoops - sorry. Should be working now, please redownload the latest release. 

Another bug I noticed is cancelling after pausing the print behaves weirdly, though that might be something I can't fix. Regular cancelling should work fine now. Could only test it three times now, my printer has some work to do for now. 

Please report back!

oopeekoo, 2019-01-14:
Quickly tested. Working as expected now.

closed 2019-01-14 by davidramiro

BlackMulch, 2019-02-11:
Love the tweaked cancellation behavior.  Is there a way to change the retraction distance upon cancellation?  For users with the v6 hotend, retraction greater than 2mm can pull hot plastic into the cold-zone... most documentation for the v6 suggest limiting retractions to 1mm or less.

I've browsed Configuration.h but I don't see any settings that scream that they're related to this.

davidramiro, 2019-02-11:
Everything related to the display menu is to be found in `AnycubicTFT.cpp`. I rarely use it myself so I'm not sure but I'm fairly sure that there is no retraction happening at all, isn't it? 

If you wanted to add retraction, you should add it on top of line 481 like this:
```
enqueue_and_echo_commands_P(PSTR("G91")); // relative mode
enqueue_and_echo_commands_P(PSTR("G1 E-1 F1800")); // retract 1mm with 30mm/s
enqueue_and_echo_commands_P(PSTR("G90")); // absolute mode
if((current_position[Z_AXIS]<150)) {
        enqueue_and_echo_commands_P(PSTR("G91"));
        enqueue_and_echo_commands_P(PSTR("G1 Z10 F240"));
        enqueue_and_echo_commands_P(PSTR("G90"));
} else if ((current_position[Z_AXIS]<190)) {
        enqueue_and_echo_commands_P(PSTR("G91"));
        enqueue_and_echo_commands_P(PSTR("G1 Z10 F240"));
        enqueue_and_echo_commands_P(PSTR("G90"));
} else if ((current_position[Z_AXIS]<200)) {
        enqueue_and_echo_commands_P(PSTR("G91"));
        enqueue_and_echo_commands_P(PSTR("G1 Z5 F240"));
        enqueue_and_echo_commands_P(PSTR("G90"));
} else if ((current_position[Z_AXIS]>=200)) {
        enqueue_and_echo_commands_P(PSTR("G91"));
        enqueue_and_echo_commands_P(PSTR("G1 Z1 F240"));
        enqueue_and_echo_commands_P(PSTR("G90"));
}
enqueue_and_echo_commands_P(PSTR("M84"));
```

BlackMulch, 2019-02-11:
Thanks for the pointer. I'll look around.

I had an OctoPrint script that retracted 1mm at cancellation, but I've disabled that. With the latest  firmware, after a canceled print, I'm needing to advance about 5mm of filament to get initial flow. I assumed this was due to the cancellation behavior (November-December firmware simply stopped, or barely moved up, upon cancellation and performed no retraction I was aware of). I'll look around. Thanks!