davidramiro/Marlin-Ai3M #10: Please add Support for I3 Mega-S Extruder

rhehl, 2019-01-20:
Please add Support for the now available I3 Mega-S Extruder:
https://de.aliexpress.com/item/ANYCUBIC-I3-Mega-Upgrade-Extruder-3D-Drucker-Kit/32970991251.html

davidramiro, 2019-01-20:
Getting that extruder to work is probably just a matter of increasing E-steps via `M92`, I heard 400 is a good starting value to calibrate from there.

As to releasing a precompiled hex file, I'd want it to be close to the manufacturer values and since I don't know what else changed besides the E-steps, I'd need to get all the acceleration and jerk settings from the Mega-S firmware. If you are running it (1.2.0 from [here](http://www.anycubic3d.com/support/show/594016.html)), please send me the output of `M503`.

rhehl, 2019-01-20:
thx, for your fast reply, i just ordered the Extruder upgrade kit.

davidramiro, 2019-01-21:
Got some info from someone with a Mega-S, I'll prepare a branch for it in the next few days.

When you mount your new extruder, start with `M92 X80.00 Y80.00 Z400.00 E384.00` and save with `M500`. Those are the default values from Anycubic. I'd still recommend calibrating it after.

ingoclaro, 2019-01-22:
I own a mega-S, before checking this I just calibrated normally as per the Readme instructions.
Something I noted is that initially it extruded very little, like 20mm, after adjusting settings I restarted the calibration process to check it's correct, and it extruded about 93mm. I input the new value and started the process again to verify, and it extruded just ~20mm again.... so finally I went with this last result and didn't check it again, but found strange that the value swings back and forth.

ingoclaro, 2019-01-23:
Here is the result of the M503 command with stock firmware:
```
Connecting...
start
Printer is now online.
echo:V1.1.2
 1.1.0-RC8
echo: Last Updated: 2016-12-06 12:00 | Author: (Jolly, xxxxxxxx.CO.)
Compiled: Jun 25 2018
echo: Free Memory: 3109  PlannerBufferBytes: 1168
echo:V27 stored settings retrieved (398 bytes)
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E384.00
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z6.00 E60.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y2000 Z60 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Z2 Endstop adjustment (mm):
echo:  M666 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P16.43 I1.04 D61.37
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:SD init fail
>>> M503
SENDING:M503
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E384.00
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z6.00 E60.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y2000 Z60 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Z2 Endstop adjustment (mm):
echo:  M666 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P16.43 I1.04 D61.37
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
Disconnected.
```

davidramiro, 2019-01-23:
Done, added a custom [branch for the Mega-S](https://github.com/davidramiro/Marlin-AI3M/tree/mega-s). Will add compiled binaries on the next release for the i3.

closed 2019-01-23 by davidramiro

cereyanli, 2020-12-14:
Has Mega S compatible compiled binaries ever published ? If yes.

Can somebody point me out to them ?

davidramiro, 2020-12-14:
I did at some point before, but as we have 3 different binaries, adding Mega-S binaries would result in 6 different binaries which is a bit much in my opinion since the Mega-S only requires a few G-Code commands to be correctly set up.