davidramiro/Marlin-Ai3M #11: [BUG] in Mega-S print hangs mid-print, screen blanks

ingoclaro, 2019-01-22:
**Describe the bug**
Printing normally and after a few minutes the printer halts and the screen in blank (all white). The only way I found to recover is to switch off the printer.

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
2. Start printing
3. Wait 5 minutes
4. Printer halts

**Expected behavior**
A clear and concise description of what you expected to happen.

**Photos of the print / output of the terminal**
how can I get the console logs/output ?

**Printer (please complete the following information):**
 - Firmware: v19.01.10 (Marlin-AI3M-190110-stock_drivers.hex)
 - Printing method: SD
 - Model: Mega-S

**M503 Output**
Please paste the complete output of the `M503` command here.

don't have this, will complete later.

davidramiro, 2019-01-23:
Thank you for your effort. So if I understand this correctly, the newest firmware shows no issue and the one you used before now causes the screen to reboot but not the board itself to hang?

To be honest, I don't see how Marlin could cause such an issue, since the display's firmware is standalone and doesn't get modified by it (the special menu being an exception, it's basically just simulating files on the SD card to show up on the screen). The screen going blank or rebooting sounds more like a hardware issue, though not even an SD card error would cause that kind of behaviour.

I'm just assuming it's a random glitch for now, since you were the only one so far seeing this issue. Keep me posted!

davidramiro, 2019-01-25:
Glad you found the cause, I hope nothing got fried when using the wrong setting!