davidramiro/Marlin-Ai3M #12: Cooling Fan Min/Max

Schildkroet, 2019-01-27:
Hello,
i replaced the cooling fans with new ones from noctua and flashed your firmware for 12V pwm output. The Hotend cooler (always 100%) runs fine, but the part cooling fan start to run about 70%+. So any values below are useless.

Is it possible to remap the pwm output without changing the source code? E.g. S1 = 70% : S255 = 100%

davidramiro, 2019-01-27:
What a coincidence, just today I received an email from a guy that had exactly the same issue, he uses Noctua fans for parts cooling as well. You cannot remap the PWM range without changing the source code.

He provided the solution: 
Uncomment `#define FAST_PWM_FAN` in `Configuration.h`. In `temperature.cpp` change line 1127 to `setPwmFrequency(FAN_PIN, 2);`

I will create a branch for it in a minute so you can easily compile it for yourself.

giacu92, 2019-01-27:
Absolutely unbelievable.. I sent David an email about the exact same issue yesterday evening.

Just a brief addition to what he said: those software modifications make the fan working on a range of about 15kHz which results in an audible noise when slowing the fan. It is not very annoying but I’m also investigating other solutions.
I have the suspicious that editing the `setPwmFrequency(FAN_PIN,2);` may somehow interfere with PID controller, but I'm currently NOT SURE and need to test it. Unfortunately I will be able to check only on next weekend.

Schildkroet, 2019-01-27:
Bugfix is working so far. Audible noise in lower frequencies is true.
Editing `setPwmFrequency(FAN_PIN,1);` solves the noise problem, but scaling of speed works now poorly. (80% of scaling is now done between 0 - 20%)

closed 2019-02-04 by Schildkroet