davidramiro/Marlin-Ai3M #15: Add M600 support

rug156, 2019-02-04:
Please add support for menu guided filament change on layer / height via M600 command

davidramiro, 2019-02-04:
I can enable everything that's needed for `M600` to work, though this command is generally meant to be used with a regular LCD display, not a custom TFT screen like the i3 Mega has. [This Marlin fork](https://github.com/systemik/Marlin/tree/1.1.x-m600-tft) has the TFT screen pause button changed to use `M600`, though I do not want to modify the default functionality on this firmware. 

If you are fine with using `M600` (pause, park nozzle and unload filament) and `M108` (load filament and resume) via serial, I can try to enable it on this firmware. If you need this to work from the screen, feel free to try out the version I linked.

rug156, 2019-02-04:
As I do not want to work with other firmwares than yours I would absolutely appreciate if you could enable M600. I have no problem using Pronterface to send serial commands to the printer.

davidramiro, 2019-02-05:
Alright, should be ready to use!

Test release download: https://kore.cc/i3mega/download/testing/m600/ 

Instructions are added [here](https://github.com/davidramiro/Marlin-AI3M/tree/m600#m600-filament-change).

[image]
Looking pretty cool I think!

davidramiro, 2019-02-05:
Glad you got it to work!

I am curious why the print didn't start until you sent `M108`. Did something appear in the terminal?

davidramiro, 2019-02-05:
Found out the reason. Pronterface - at least for me on the latest macOS version - constantly sends out `M105` to check the temperature instead of using `M155` (temperature auto-reporting) which is the recommended way of doing it. Sending `M105` while the printer is heating up with `M109` causes the firmware to halt and wait for user input for some reason. I'll investigate further tomorrow. 

Thanks for mentioning this, I would never have noticed!

lkspencer, 2019-02-06:
I'm not sure if any of this in intended behavior (I'm new to 3d printing), so I wanted to point out my experience with this recent addition.

I started a print and then in the middle of the print send `M600` via pronterface. It paused the print, moved the print head and beeped (can't remember the exact order those happened in) but it did not retract the filament. If I send an `M600 B4 U-578` then it will retract the filament to right about the start of the bowden tube, but it beeps after retracting and not before.

In either of the cases above, when I have changed my filament and I issue the `M108` command via pronterface, it only inserts the filament about 30mm into the tube far from the actual nozzle and the print won't work because no filament will get extruded for about another 558mm.

rug156, 2019-02-06:
re-read the readme. you did it wrong.

davidramiro, 2019-02-06:
I'm not very familiar with the pause function myself, but as I understand it, since `M108` has no parameters and simply breaks out of the pause, you'd have to specify the load length before. Marlin docs say `M600 L578 U578` would work like you intended, though I didn't try. 
What you were seeing after `M108` was simply a purge:
```
#define ADVANCED_PAUSE_PURGE_LENGTH         50  // (mm) Length to extrude after loading.
```

What I did try was setting the parameters in EEPROM via `M603`. Defining the length before resulted in the correct loading and unloading with an `M600`, no parameters needed then of course.

That being said, M600 is and was always intended to be used with a generic Marlin LCD screen that allows the user to better control the behaviour. Headless mode like we are using it is not a very elegant way, though it was luckily included in recent changes.

davidramiro, 2019-02-06:
@rug156 
I just fixed the issue where the print doesn't start until you send `M108`.

It was a weird rx buffer overflow which caused an `M105` command get sent as `M1` - which is a GCode to pause the printer.

I'll add the fixed release in a few minutes.

rug156, 2019-02-06:
thx. will test in the evening.

lkspencer, 2019-02-06:
@rug156, thank you. I did miss the configuration step in the readme.

@davidramiro you have done a fantastic job with this firmware. Thanks for all your hard work!

davidramiro, 2019-02-06:
Thank you! Got a bunch of helpful suggestions and feedback like this. I would have never known that M600 is that easy to use.

I will now try to get the unload and load/purge lengths right for the bowden tube. 578mm was too long for my taste, a quick test with 560mm is almost where I want it to be.

davidramiro, 2019-02-09:
After some more testing, I'll consider this feature working. ☺️

Feel free to provide further feedback and suggestions.

closed 2019-02-09 by davidramiro