davidramiro/Marlin-Ai3M #16: Interval beeping on second layer [BUG]

Trader2k, 2019-02-04:
Description:
An beeping (high frequency, but quiter than the regular start signal) if the code is reaching the second layer. video (link to youtube) follows tomorrow.

ANY I3 Mega

MODs:
Only regular things, like: 
- Hotend fan (noctua)
- board  and psu fan (Sunon Lüfter 60x60x15mm HA60151V4-999)
- stepper tmc2208

and since yesterday Marlin 1.1.9

davidramiro, 2019-02-05:
I'm curious to see a video since I can't really imagine what's going on. If it's a continuous high frequency humming noise, it might come from the parts cooling fan.

Trader2k, 2019-02-05:
https://youtu.be/YdNU2q8EwUQ

davidramiro, 2019-02-05:
Does it stop when you manually stop the fan via display while on the second layer? I am almost sure it's the fan making this noise. 

If so, I am almost certain that this is due to Curas fan settings. If you cap the maximum fan speed at 60%, Cura will try to start the fan with like 12% power on the second layer. This causes the fan not to spin and make this noise.

I don't usually use Cura, but setting the Initial Fan Speed setting to 30% should help. Something like this would be my recommendation:

[image]

davidramiro, 2019-02-05:
>I don't think so. The fans are silent and another friend has the same problem. He has stock fans and I have extra silent.

I am talking about the parts cooling fan, not the other ones. It's also the only one that is controlled via GCode, which explains why it starts on the second layer - where cooling kicks in.

Trader2k, 2019-02-06:
You are correct. Now, everything is nice, Thx for the good and fast support. 
You can change label of this issue to "Layer 8 fault" ;-)

Greetings

closed 2019-02-06 by davidramiro