davidramiro/Marlin-Ai3M #19: Add support for BLTouch

matzebecker, 2019-02-06:
Hi davidramiro,

I really like the Bowden MK3 R3 X-carriage upgrade for the i3 Mega [1] that comes with a BLTouch sensor. Adding support for this sensor would be great! A firmware for the i3 Mega with BLTouch support can be found here [2].

PS: Thanks for the great work!

[1] https://www.thingiverse.com/thing:3092917
[2] https://drive.google.com/drive/folders/1KSmF20qrJIhvWThyRuXINJi3H0W_XMkA

davidramiro, 2019-02-06:
Hi Matthias,

There is probably no need for that, since there is already a fork of this firmware will full BLtouch support including a detailed Wiki with instructions on how to wire and adjust it:
https://github.com/DerDominik/Marlin-AnycubicI3Mega-BLTouch

Furthermore, I am hesitant to include features I cannot test myself. I am sure this firmware will work out great for you, I can also forward any questions or issues to the developer since I am in touch with him. 

Using that fork you could also always easily merge any updates from this firmware.

matzebecker, 2019-02-06:
Thank you! I wasn't aware of the fork. Thanks again for all your great work!

closed 2019-02-06 by davidramiro

Krimik, 2019-02-11:
@matzebecker, did you manage to get it working? I am struggling with my Anycubic i3 and Bltouch for many days now and it just does not want to work with that fork...

matzebecker, 2019-02-12:
@Krimik Not yet. I didn't have the time to install the BLTouch yet. I'll let you know as soon I did manage to get everything working.