davidramiro/Marlin-Ai3M #20: Update for v19.01.22 ?

alienfactor, 2019-02-09:
Hello David,
is there a pwm-version available for the latest v19.02.08 stable release?

Thanks and regards

davidramiro, 2019-02-09:
I will compile the latest stable with the PWM settings when I get home and post the download link here.

I won't do those regularly since there are many different branches, but I will write some instructions on how to edit the newest releases yourself for settings like software PWM later today.

alienfactor, 2019-02-09:
Fast as lightning :)
Thank you very much!

davidramiro, 2019-02-09:
https://kore.cc/i3mega/download/releases/special/soft-pwm/

closed 2019-02-09 by davidramiro

rug156, 2019-02-09:
Sorry for asking noob questions, but what is this PWM for?

davidramiro, 2019-02-09:
@rug156 
It's a certain way of powering the fan. Needed for some particular fans like from Sunon. Not needed on the stock one.

alienfactor, 2019-02-10:
@rug156
https://en.wikipedia.org/wiki/Pulse-width_modulation

Finally i realized, that my Sunon MF50151VX-A99 isn't a PWM fan (it's stuttering with pwm firmware).
Maybe it's a damaged one, because at low rotations you can hear an anoying humm/buzz..

davidramiro, 2019-02-10:
I'd recommend you to look into editing and compiling the firmware yourself and try out some changes. A rough explanation on how to compile with Arduino IDE is on the readme.

There is no F-parameter on Marlin's M106. Refer to [Marlin's documentation](http://marlinfw.org/docs/gcode/M106.html), other GCode docs might list other firmare's capabilities.

In `Configuration.h`, uncomment `#define FAN_SOFT_PWM` and set `#define SOFT_PWM_SCALE` to 2. That works for some fan models. If it doesn't work for you, revert the changes from before and try uncommenting `#define FAST_PWM_FAN`.