davidramiro/Marlin-Ai3M #21: [HELP?] Hotend fan won't turn off

chrisnicola, 2019-02-09:
**Describe the bug**
This may not be an issue with the firmware but my hope is the experts around here might be able to lend me a hand. I installed the firmware on my upgraded i3 Mega which has a Titan Aero hot end. The heatsink fan used to turn off when the hot end temperature was below a certain point (around 40c or something like that).

When I installed the Aero I missed the instruction about not over tightening the thermistor cartridge and it was damaged somewhat, but seemed to work fine for quite a while. At some point, it failed in a way that caused the hot end to overhead to the point of burning the filament to a crisp (no fire though and I caught it).

I was able to confirm the thermistor was shot and ordered a replacement which I just got and installed correctly this time. It is reporting temperature correctly now, but for some reason as long as the machine is turned on the heatsink fan is always running now. Nothing will shut it down.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on printer
2. Thermistor temp reading 20 degrees
3. Fan is running

**Expected behavior**
Fan should turn off when the hot end is not heated.

**Printer (please complete the following information):**
 - Firmware: [e.g. 181227]
 - Printing method: Typically Curae[ep
 - Model i3 Mega with Ultravase

**M503 Output**
Please paste the complete output of the `M503` command here.

```
09:24:40.167 : N18 M503*28
09:24:40.173 : echo:  G21    ; (mm)
09:24:40.173 : echo:Filament settings: Disabled
09:24:40.173 : echo:  M200 D1.75
09:24:40.173 : echo:  M200 D0
09:24:40.173 : echo:Steps per unit:
09:24:40.176 : echo:  M92 X80.00 Y80.00 Z400.00 E405.00
09:24:40.176 : echo:Maximum feedrates (units/s):
09:24:40.180 : echo:  M203 X500.00 Y500.00 Z6.00 E60.00
09:24:40.180 : echo:Maximum Acceleration (units/s2):
09:24:40.184 : echo:  M201 X3000 Y3000 Z60 E10000
09:24:40.184 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
09:24:40.192 : echo:  M204 P1000.00 R3000.00 T3000.00
09:24:40.192 : echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
09:24:40.213 : echo:  M205 Q20000 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
09:24:40.213 : echo:Home offset:
09:24:40.213 : echo:  M206 X3.00 Y-10.00 Z0.00
09:24:40.213 : echo:Mesh Bed Leveling:
09:24:40.213 : echo:  M420 S0 Z0.00
09:24:40.213 : echo:  G29 S3 X1 Y1 Z0.04000
09:24:40.213 : echo:  G29 S3 X2 Y1 Z0.08000
09:24:40.213 : echo:  G29 S3 X3 Y1 Z0.08000
09:24:40.213 : echo:  G29 S3 X4 Y1 Z0.02000
09:24:40.213 : echo:  G29 S3 X5 Y1 Z-0.02000
09:24:40.213 : echo:  G29 S3 X1 Y2 Z0.00000
09:24:40.213 : echo:  G29 S3 X2 Y2 Z0.06000
09:24:40.213 : echo:  G29 S3 X3 Y2 Z0.08000
09:24:40.213 : echo:  G29 S3 X4 Y2 Z0.04000
09:24:40.213 : echo:  G29 S3 X5 Y2 Z-0.04000
09:24:40.213 : echo:  G29 S3 X1 Y3 Z0.00000
09:24:40.213 : echo:  G29 S3 X2 Y3 Z0.06000
09:24:40.231 : echo:  G29 S3 X3 Y3 Z0.08000
09:24:40.231 : echo:  G29 S3 X4 Y3 Z0.02000
09:24:40.231 : echo:  G29 S3 X5 Y3 Z-0.04000
09:24:40.231 : echo:  G29 S3 X1 Y4 Z0.00000
09:24:40.231 : echo:  G29 S3 X2 Y4 Z0.06000
09:24:40.231 : echo:  G29 S3 X3 Y4 Z0.06000
09:24:40.231 : echo:  G29 S3 X4 Y4 Z0.02000
09:24:40.231 : echo:  G29 S3 X5 Y4 Z-0.06000
09:24:40.231 : echo:  G29 S3 X1 Y5 Z0.00000
09:24:40.231 : echo:  G29 S3 X2 Y5 Z0.04000
09:24:40.231 : echo:  G29 S3 X3 Y5 Z0.06000
09:24:40.231 : echo:  G29 S3 X4 Y5 Z0.02000
09:24:40.231 : echo:  G29 S3 X5 Y5 Z-0.04000
09:24:40.231 : echo:Endstop adjustment:
09:24:40.231 : echo:  M666 Z0.00
09:24:40.231 : echo:PID settings:
09:24:40.231 : echo:  M301 P31.96 I3.61 D70.68
09:24:40.234 : echo:  M304 P188.98 I33.93 D263.18
09:24:40.234 : echo:Linear Advance:
09:24:40.234 : echo:  M900 K0.06
```

davidramiro, 2019-02-09:
Hi, 

sadly I am almost sure this is a hardware issue. The firmware definitely turns off the hotend fan at <50°C, as seen in Configuration_adv.h, line 246:
`#define EXTRUDER_AUTO_FAN_TEMPERATURE 50`

The transistor that controls the hotend fan might have went bad, maybe when the thermistor shorted out?

As a dirty workaround, you could install a Noctua NF-A4x10 FLX, it fits perfectly and is almost silent. Otherwise you will probably have to replace your board...

chrisnicola, 2019-02-09:
@davidramiro thanks for the reply. So I updated the firmware to the latest just to be sure and same behaviour.

However I just noticed something odd. The fan is louder when the hotend is cool and then it slows down and gets quiet when it goes above 50c. So it seems like it is putting out the wrong voltage when it is supposed to be off.

Any idea what might cause that.

davidramiro, 2019-02-09:
That points even more towards a blown transistor :worried: 
Just to make sure it's a hardware fault, you could go back to the manufacturer firmware.

chrisnicola, 2019-02-09:
So when the temperature is above 50c the voltage output is 8.36v but when it drops below that it changes to 12.84v. Definitely something funny hardware wise.

Any ideas on what needs to be replaced to fix that or does the entire board need to be scrapped?

davidramiro, 2019-02-09:
Might very well be this transistor here:

[image]

If you're comfortable with SMD soldering, I can look up possible replacements though the whole board is only like $30. Also maybe the Anycubic support even grants you a free replacement.

chrisnicola, 2019-02-09:
Thanks I've contacted support. Probably best to just replace the whole board. If I'm feeling impatient I can probably get one and the electronics store down the street.

davidramiro, 2019-02-09:
Since this isn't really related to this firmware, I'm closing this issue for now.

I'd recommend you to post this question to the i3 Mega Facebook or Thingiverse groups. Plenty of helpful and knowledgeable people on there.

closed 2019-02-09 by davidramiro

chrisnicola, 2019-02-09:
Thanks for all your help @davidramiro!