davidramiro/Marlin-Ai3M #22: OctoPrint/AstroPrint USB Connection abort

alienfactor, 2019-02-09:
Hi David,

with the current/latest release i have serious connection problems.
Printing from SD works fine, but as soon i use an external USB controller (AstroPrint) the connection get lost after a few minutes.
Right now its not possible to print anything else, except from SD card.

I hope its not a problem with my printer itself..

davidramiro, 2019-02-09:
Weird, I do not see these issues. Can you provide `octoprint.log` and `serial.log`?

alienfactor, 2019-02-10:
The issue was only with with Astropring (a branch of Octoprint).
I switched back to Octopring .. and it works!
So my bad, it just don´t work with that (too expensive) piece of .. Software

Sad because i really like the UI on a larger Screen, but i can deal with that :)

davidramiro, 2019-02-10:
As AstroPrint is based on OctoPrint, maybe it also has some log functionality? I can try to look into it.

alienfactor, 2019-02-10:
I will check it later on. If there are logs, i can find them :)
P.S.: There are different versions. A free one and a payed (touch) version (incl. beta versions). Not sure if the are identical..

davidramiro, 2019-02-13:
Any new info? Found this page: https://astroprint.zendesk.com/hc/en-us/articles/115001797403-Why-does-my-printer-disconnect-in-the-middle-of-a-print-

If it's indeed the firmware, the last point describes how to enable serial logs.

alienfactor, 2019-02-13:
Will test it (hopefully) today and post the result/log.

The problem was not during the print process, it was before the start (abort after heatup).

davidramiro, 2019-02-13:
Ah - might already have been fixed in bfe39a342dcdf8881ad221dab6ed22a169af86b7 and the latest release. There was a similar issue with USB printing via Simplify3D.

alienfactor, 2019-02-16:
Hello David,

you`re right. I tested it again and it seems to work flawlessly now :)

closed 2019-02-16 by davidramiro