davidramiro/Marlin-Ai3M #23: Heat bed can not be fully used

michinator, 2019-02-11:
Hello,

no idea if it is a bug. After switching from Marlin 1.1.0 to 190208-TMC2208, the heating bed max can only be heated up to 105 ° C.

The PID tuning is unfortunately not possible at 110°. The error message: "PID Autotune failed! Temperature too high" is displayed in the octo print terminal.

Otherwise, the new pid function works great. There are two mosfets installed in the i3 mega v3. I hope you can help me, because I print mostly abs at 110 °.

here are the calculated optimized values from the bed

Recv: #define DEFAULT_bedKp 241.85
Recv: #define DEFAULT_bedKi 47.62
Recv: #define DEFAULT_bedKd 307.09

Thanks in advance

davidramiro, 2019-02-11:
Confirmed. Since the latest merge of Marlin's official repo (f01ee750049e30b66c257c9c4cb0f82278236f61) heating only goes up to `MAXTEMP` minus 15°C. I'll adjust them later today.

michinator, 2019-02-11:
👍

davidramiro, 2019-02-11:
Done, check the newest release. New maximum temps are 275°C and 125°C.

michinator, 2019-02-11:
try it now
great support :)

michinator, 2019-02-11:
it works.
Thanks again

closed 2019-02-11 by michinator