davidramiro/Marlin-Ai3M #25: pause dos'ntwork during printing. nozlle won't lift just stays on same layer .  [BUG]

MD3DDelta, 2019-02-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
2. Move axis by '....'
3. Start printing '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Photos of the print / output of the terminal**
If applicable, add photos or console outputs to help explain your problem.

**Printer (please complete the following information):**
 - Firmware version: [e.g. v19.02.08]
 - Printing method [e.g. SD, USB (OctoPrint, Cura, ...)]
 - Model [e.g. i3 Mega, Mega-S]

**M503 Output**
Please paste the complete output of the `M503` command here.

davidramiro, 2019-02-16:
Confirmed, working on it on the dev branch.

https://github.com/davidramiro/Marlin-AI3M/tree/dev

Will update in a few days.

MD3DDelta, 2019-02-16:
Wow , you are the best , thanks .

davidramiro, 2019-02-16:
So far I reworked the handling of filament runout and pause here: https://github.com/davidramiro/Marlin-AI3M/commit/1fe1842a75ea57362bc6225d3ac3674300937dd5

For now, using pause raises the nozzle 10mm and retracts 7mm of filament.

I don't remember how the manufacturer firmware handles pause, could you please add that to the expected behaviour of the bug report template? (it's meant to be filled out so, see your first post and edit it accordingly please)

MD3DDelta, 2019-02-17:
thanks david you are awesome , to prevent oozing , is very helpfull.
Ii  use the pause for unglog the nozzle , i have an few spools of old filament aand often they are bad pieces in them. every splool is dry and clean , just old about 5 years . 
How ca i support you with some money , you have defently earend it . 
paypal or wath else ?
thank you so much . 
when i end my upgrade on the mega i3 x-cariage i let you know . 
cheers mike

davidramiro, 2019-02-17:
The latest release candidate can be found here:
https://kore.cc/i3mega/download/testing/pause/latest/

Has been tested a couple of times on two different machines, seems to work fine so far. Please check it out for yourself and report back.

And yes, I do have [PayPal](https://paypal.me/djusto/). A donation is not expected but of course always highly appreciated. 😀

MD3DDelta, 2019-02-18:
it works , Just almost like the orginal .
Only 1 motor is winey, when it paused . i could't find out wich one . But the steppers stay cool , so is it a bit more power on .
i just send you 10 € , thank you for your work .
Regards Mike

closed 2019-02-18 by MD3DDelta