davidramiro/Marlin-Ai3M #28: [BUG] Printer Menu Freezes on SD Card after flashing

Escapado, 2019-02-21:
I just received my Anycubic I3 mega. It had Firmware 1.1.0 on it. I flashed the custom firmware using Cura 3.6.0. I can navigate all the menus just fine. Once I plug in an SD Card the Print menu shows the following options:

> <Special_Menu>
> Owl_pair.gcode
> /Files_Engish_Mega
> /D™_-‡_Mega

However the touch screen doesn't react and I can't navigate the menu anymore. I can use Pronterface and move the motors so its not entirely unresponsive. To get out of the menu I have to restart the printer. Once I enter it again, it freezes again. Down below you'll find an image of what I see and where it freezes and the M503 Output.

https://drive.google.com/file/d/1YiHG0ZxP2h58uPbw5SjJnTwriLZKem5f/view?usp=sharing

M503 Output
```
SENDING:M503
echo:  G21    ; (mm)
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E92.60
echo:Maximum feedrates (units/s):
echo:  M203 X500.00 Y120.00 Z6.00 E60.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X2000 Y1200 Z60 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1200.00 R3000.00 T1500.00
echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 Q50000 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Mesh Bed Leveling:
echo:  M420 S0 Z0.00
echo:Endstop adjustment:
echo:  M666 Z0.00
echo:PID settings:
echo:  M301 P15.94 I1.17 D54.19
echo:  M304 P251.78 I49.57 D319.73
echo:Linear Advance:
echo:  M900 K0.00
echo:Filament load/unload lengths:
echo:  M603 L538.00 U555.00

```

davidramiro, 2019-02-21:
Hi there!
Right from the readme:

> Special characters on any file or folders name on the SD card will cause the file menu to freeze. Simply replace or remove every special character (Chinese, Arabic, Russian, accents, German & Scandinavian umlauts, ...) from the name. Symbols like dashes or underscores are no problem. 
> **Important note: On the SD card that comes with the printer there is a folder with Chinese characters in it by default. Please rename or remove it.**

Guess you missed that part. Delete everything but the gcode from that SD card and try again 😊

Escapado, 2019-02-21:
Damn, I really missed that. I am such a dork. Thanks! :)

closed 2019-02-21 by Escapado

Arut42, 2019-12-14:
have the same problem on 2.0 what did u change?
best regards

davidramiro, 2019-12-14:
>Special characters on any file or folders name on the SD card will cause the file menu to freeze. Simply replace or remove every special character (Chinese, Arabic, Russian, accents, German & Scandinavian umlauts, ...) from the name. Symbols like dashes or underscores are no problem.
Important note: On the SD card that comes with the printer there is a folder with Chinese characters in it by default. Please rename or remove it.