davidramiro/Marlin-Ai3M #29: Abort print via LCD leads to a movement of the hotend into the printed part [BUG]

SimGas, 2019-03-06:
Yesterday I started a print and everything was just working fine. At a height of approx 26mm I got some issues with the extruder which are most likely based on a misconfiguration of my slicing software.

However, when I aborted the print using the LCD "STOP"-Button, the printer stopped printing and moved the Z-Axis down right into the print. 

I guess there is some problem with the stop code, which moves the printhead to an absolute position instead of a relative plus.

-Firmware version: davidramiro - v1.4.0
-Printing method: SD
-Model: Mega-S

Newbie-2, 2019-03-06:
Why do not you use the version v1.4.2 but the old v1.4.0 ?

davidramiro, 2019-03-06:
Confirmed. Had a similar bug report a few days ago. 

Relative/absolute positioning seems plausible, good idea. That `G91` line seems to get lost somehow since it's definitely in the code. I'll try flushing the buffer before.

SimGas, 2019-03-06:
> Why do not you use the version v1.4.2 but the old v1.4.0 ?

Good question: I just don't want to reconfigure my firmware every time a new version gets released. 
But you are definitely right. Before submitting a bug report, I should`ve checked if the problem still occurs with a new firmware version.

Anyways, david could still reproduce the bug with the newest firmware.

davidramiro, 2019-03-06:
Think I got it sorted out. Feel free to compile and test, otherwise I'll add a new release tomorrow.

Also, reconfiguring is not needed on most of the updates. Just flash and go ahead. If it's a major update that needs an EEPROM reset, I'll always mention it in the release notes.

SimGas, 2019-03-07:
Awesome, thanks!

I'll check it out this evening.

davidramiro, 2019-03-08:
While the hotend does indeed not move downwards anymore, sometimes it now wouldn't move up now since the command after `G91` got lost (not repeatable). Seems like `enqueue_and_echo_commands_P` doesn't like chained commands.

I think this fixes it for good, tested 5 times now:
```
void AnycubicTFTClass::StopPrint(){

  #ifdef SDSUPPORT
    card.stopSDPrint();
  #endif
  clear_command_queue();
  quickstop_stepper();
  print_job_timer.stop();
  thermalManager.disable_all_heaters();
  enqueue_and_echo_commands_P(PSTR("G91"));
  if((current_position[Z_AXIS]<150))
  {
    enqueue_and_echo_commands_P(PSTR("G1 Z20 F240"));
  }
  else if ((current_position[Z_AXIS]<190))
  {
    enqueue_and_echo_commands_P(PSTR("G1 Z10 F240"));
  }
  else if ((current_position[Z_AXIS]<200))
  {
    enqueue_and_echo_commands_P(PSTR("G1 Z5 F240"));
  }
  else if ((current_position[Z_AXIS]>=200))
  {
    enqueue_and_echo_commands_P(PSTR("G1 Z1 F240"));
  }
  enqueue_and_echo_commands_P(PSTR("G90"));
  #if FAN_COUNT > 0
    for (uint8_t i = 0; i < FAN_COUNT; i++) fanSpeeds[i] = 0;
  #endif
  TFTstate=ANYCUBIC_TFT_STATE_SDSTOP_REQ;
  enqueue_and_echo_commands_P(PSTR("M27")); // force report of SD status
}
```

davidramiro, 2019-03-11:
Scratch that. I have completely reworked the code to park the nozzle properly in 10369de0aba2438a42e7d8575d5c08bbdb778e29. Now it raises Z by 25mm, moves the bed forward for easy removal and retracts 2mm. No Gcode used, no relative positioning needed. All done in native functions. 

That should solve this issue for good. I'll release the binaries tomorrow.

davidramiro, 2019-03-13:
Fixed in v1.4.4.

closed 2019-03-13 by davidramiro