davidramiro/Marlin-Ai3M #30: [BUG]

tis-cs, 2019-03-11:
Hi, erstmal ein ganz großes Lob für deine Arbeit! Ich finde es absolut super, was hier alles möglich gemacht wird! 

Mir ist beim Autobedleveling aufgefallen, dass die 0.02er Schritte nicht wirklich laufen. Ich habe mehrfach 0.02er Schritte ausgeführt und dennoch hat sich die Z-Achse nicht bewegt. Wenn ich jedoch einen 0.1er Schritt mache, geht es sofort, was jedoch in einigen Fällen zu viel war.

Ich hatte manchmal das Gefühl (absolut subjektiv) dass er nach mehrmaligem hoch- und runterfahren von 0.1er Schritten dann irgendwann auch mal einen 0.02er genommen hat.

Meine Erwartung wäre, dass wenn ich fünf Mal einen 0.02er Schritt in eine Richtung mache, dass es das selbe Ergebnis wäre wie ein 0.1er Schritt.

```
Recv: echo:Marlin 1.1.9
Recv: Marlin-AI3M v1.4.2
Recv: echo: Last Updated: 2019-02-24 | Author: (davidramiro)
Recv: echo:Compiled: Feb 24 2019
Recv: echo: Free Memory: 2186  PlannerBufferBytes: 1664
Recv: echo:V55 stored settings retrieved (731 bytes; crc 42887)
Recv: FIRMWARE_NAME:Marlin 1.1.9(Github, davidramiro) MACHINE_TYPE:Anycubic i3 Mega EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff MARLIN-AI3M_VERSION:v1.4.2
Recv: FIRMWARE_NAME:Marlin 1.1.9(Github, davidramiro) 
```

```
Send: M503
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E398.50
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y120.00 Z6.00 E60.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X2000 Y1200 Z60 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1200.00 R3000.00 T1500.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 Q50000 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:  G29 S3 X1 Y1 Z0.00000
Recv: echo:  G29 S3 X2 Y1 Z0.02000
Recv: echo:  G29 S3 X3 Y1 Z0.02000
Recv: echo:  G29 S3 X4 Y1 Z-0.06000
Recv: echo:  G29 S3 X5 Y1 Z-0.16000
Recv: echo:  G29 S3 X1 Y2 Z-0.06000
Recv: echo:  G29 S3 X2 Y2 Z0.02000
Recv: echo:  G29 S3 X3 Y2 Z0.02000
Recv: echo:  G29 S3 X4 Y2 Z-0.06000
Recv: echo:  G29 S3 X5 Y2 Z-0.16000
Recv: echo:  G29 S3 X1 Y3 Z-0.06000
Recv: echo:  G29 S3 X2 Y3 Z0.02000
Recv: echo:  G29 S3 X3 Y3 Z0.02000
Recv: echo:  G29 S3 X4 Y3 Z-0.04000
Recv: echo:  G29 S3 X5 Y3 Z-0.18000
Recv: echo:  G29 S3 X1 Y4 Z-0.08000
Recv: echo:  G29 S3 X2 Y4 Z0.02000
Recv: echo:  G29 S3 X3 Y4 Z0.04000
Recv: echo:  G29 S3 X4 Y4 Z-0.06000
Recv: echo:  G29 S3 X5 Y4 Z-0.18000
Recv: echo:  G29 S3 X1 Y5 Z-0.10000
Recv: echo:  G29 S3 X2 Y5 Z-0.02000
Recv: echo:  G29 S3 X3 Y5 Z-0.02000
Recv: echo:  G29 S3 X4 Y5 Z-0.02000
Recv: echo:  G29 S3 X5 Y5 Z-0.16000
Recv: echo:Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P15.94 I1.17 D54.19
Recv: echo:  M304 P251.78 I49.57 D319.73
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L538.00 U555.00
Recv: ok
```

Danke!

davidramiro, 2019-03-11:
Hi,

ich gehe jetzt einfach mal davon aus, dass du die Z-Achse über das Display ansteuerst. Das Special Menu ist nicht so gut für wiederholtes Senden von Befehlen geeignet - man muss für jeden Befehl den Menüpunkt nochmals auswählen und danach erst den runden Pfeil drücken, siehe Anleitung: 

> (If you want to send the same command multiple times, select the item again, even though it is still marked red.)

Um zu bestätigen, dass es am Menü lag, kannst du auch mal versuchen, die Z achse per Pronterface, OctoPrint, o.Ä. anzusteuern:

> Manual commands for use with OctoPrint etc.:
> 
> - `G29 S1` - Start MMBL
> - `G29 S2` - Next Mesh point
> - Raising Z: `G91`, `G1 Z0.02`, `G90` (one after another, not in one line)
> - Lowering Z: `G91`, `G1 Z-0.02`, `G90`
> - After seeing ok in the console, send `M500` to save.

Sollte das auch nicht funktionieren, liegt hier eher ein mechanisches Problem vor.

tis-cs, 2019-03-11:
> man muss für jeden Befehl den Menüpunkt nochmals auswählen und danach erst den runden Pfeil drücken

das würde natürlich erklären, warum es nach einem 0.1er Schritt mal ging... Somit lag der Fehler bei mir und ich habe die Anleitung an der Stelle überlesen. Sorry dafür und danke für deine Hilfe!

Somit nur ein nice to have, dass man es nicht pro Schritt neu anwählen muss :)

davidramiro, 2019-03-11:
Leider schwer umsetzbar, das Menü ist ja eigentlich gar nicht dafür gedacht sondern soll nur Dateien auflisten die einmal angewählt werden 🙂 

Das Mesh Leveling muss man aber eh recht selten machen, sollte also kein allzu großer Umstand sein.

closed 2019-03-11 by davidramiro

badabing2005, 2019-05-12:
> Sollte das auch nicht funktionieren, liegt hier eher ein mechanisches Problem vor.

First of all kudos for an awesome job, the printer is so much better with this firmware.
Secondly sorry for writing in English, I don't know any German and I used Google translator to understand what was discussed in this thread as I seeked a solution for my problem.

You suggested that if that doesn't work it's probably a mechanical problem.
I followed the instructions, and always first re-selected 0.02 move before hitting the circle.
What my issue is that if I do a 0.02 move up, and then follow it with a 0.02 move down (making sure to select properly and hit circle) it doesn't necessarily go back to the same height (sometimes it does sometimes it does not)
Of course I'm judging this based on the resistance I feel against the paper
Sometimes one 0.02 up makes it very loose, and then I have to follow it with 2, 3, or even 4 0.02 down to bring it to a properly resistance.

What could the issue be, if it is mechanical, what part need I change?
If indeed I have a mechanical issue, what print artifacts / flaws should I observe?

Thanks and have a good day.
BB

davidramiro, 2019-05-13:
@badabing2005 

Well, the repeatability of such small steps is just not very good on this printer, so the inaccuracy is not a flaw on your machine but a general issue.

To somewhat increase the accuracy, you could move up Z by 0.1mm first (which should be fairly accurate) and then move Z down by 0.2mm 4 or 6 times (to get a net movement of -0.02 mm or +0.02 mm). I think downwards movement is better for small movements because of less torque being required.

Also, you don't have to be perfect on the leveling. As long as you aren't crushing your paper or not touching it at all, you can compensate small inconsistencies by increasing first layer height/width in your slicer.

badabing2005, 2019-05-14:
Thank you @davidramiro 
I'm relieved to hear that it's not a flaw of my machine.
If downward movement is relatively more accurate, would it help if I tune with the wheels in such a way that all I would need is a downward movements to fine tune with auto-leveling?
I was attempting to tune with the wheels to a level that I would be able to fine tune with moving upward or downwards.
I guess if I find the lowest point, and tune that to be accurate (with the wheels) then I adjust all the others with only downward movements.
I find 0.1 move is quite large, as either crushes to doesn't touch the paper hence why I was using 0.02 movements.

On another note is that a part that could be changed that could improve the z axis precision?
Perhaps the stepper motor or the drivers?
Sorry for silly questions, I'm new to this, and thanks for taking the time to respond.

Cheers
BB