davidramiro/Marlin-Ai3M #31: [FR] Reduced retraction multiplier during Mesh Validation

BlackMulch, 2019-03-24:
**Describe your suggestion.**
I recommend changing the default [retraction multiplier for G26 Mesh Validation](https://github.com/davidramiro/Marlin-Ai3M/blob/79f559e9aff8f605b6ca20ba195eb8c41142fd48/Marlin/G26_Mesh_Validation_Tool.cpp#L47) to a lower value:  I recommend a value of 1.

**Reasoning**
The current value results in a default retraction of 6mm and unretraction of 7.5mm during the mesh validation print:  This is a lot of retraction and priming, even for a stock i3 Mega.  A lower value would be less likely to pull molten plastic out of the melt zone, and result in more consistent line widths during the mesh validation print.  One possible drawback is that there is more likely to be oozing and stringing on this print, but I'd rather have that than a jammed hotend.

**References, context or examples.**
The retraction multiplier can also be manually set by passing the Q parameter with the [G26 mesh validation command](http://marlinfw.org/docs/gcode/G026.html), so an alternative to adjusting the default value is to update the readme pointing out this parameter.  Basic Marlin (not customized for the Mega) uses a [default of 1](https://github.com/MarlinFirmware/Marlin/blob/5befb307011607cc5ff391d974f2087e728c6099/Marlin/G26_Mesh_Validation_Tool.cpp#L47) for this value.

davidramiro, 2019-03-24:
Hi there and thanks for the suggestion!

Agreed. To be honest, I was never very happy with `G26` myself. 

>The current value results in a default retraction of 6mm and unretraction of 7.5mm during the mesh validation print: This is a lot of retraction and priming, even for a stock i3 Mega.

Actually, 6mm retractions _can_ be fine on the stock hotend. In my opinion, the bigger issue here is the fixed multiplier of 1.2, trying to feed 1.5mm more than we retracted was less than ideal of course.

>I recommend a value of 1.

That might work well on your v6 hotend (I've seen your fork 🙂) but the stock one definitely needs some more retraction. 

No big deal though, I've made some changes to the G26 Gcode. You can now set retraction and unretraction independently from each other. I still preserved Marlin's defaults if nothing is specified, as pointed out in the code:

```
   *   Q #  Retract     Retraction length. Defaults to 1mm if not specified.
   *   Z #  Unretract   Unretraction length. Defaults to 1.2mm if not specified.
   *                    Note: If Q is specified but Z isn't, Z defaults to Q * 1.2.
```

So you can now either adjust your settings with the `RETRACTION_LENGTH` and `UNRETRACTION_LENGTH` definitions in `G26_Mesh_Validation_Tool.cpp` or via `Q` and `Z` parameters on the fly.

As for the stock hotend, I found `Q4.2 Z4` to work well. A longer feed than retract just leads to unnecessary stringing. Line width is still consistent. Edited the readme accordingly.

BlackMulch, 2019-03-25:
Agreed. Thanks for taking a look. 4mm of retraction was about what I was using before I switched to the v6 (which is what moved me to your excellent firmware!).

Thanks again for all your hard work.

closed 2019-03-26 by davidramiro