davidramiro/Marlin-Ai3M #32: [FR] Marlin 2 porting

iherbak, 2019-04-04:
**Describe your suggestion.**
Trying to integrate into marlin 2.0 bugfix.
Have a version which compiles, but the LCD is not working. I could provide you the modified source.

Could you assist me with that?

Thanks

**Reasoning**
Would be great to have the LCD working in the newest version as well
[Marlin.zip](https://github.com/davidramiro/Marlin-Ai3M/files/3043127/Marlin.zip)

davidramiro, 2019-04-04:
https://github.com/davidramiro/Marlin-Ai3M-2.0.x

Already done 🙂 

I decided to move this to another repo where I can easily merge all the changes from upstream. Feel free to take a look around over there.

iherbak, 2019-04-04:
HI!

Thanks for the answer. Havent noticed your branch.

Just uploaded it, it works.Now i can look around what i did wrong in my
version.

Best regards,
István

davidramiro, 2019-04-04:
From a quick glance, while the display code does compile fine on your version, it was just some hooks into the Marlin code that were missing. Things like `AnycubicTFT.Setup()` in the `Marlin.cpp` setup routine or `AnycubicTFT.CommandScan()` in loops like `idle()`. So you were pretty close, the code just didn't get called 🙂

iherbak, 2019-04-04:
HI,

Thanks a lot. I'm glad i was close :)
It was a 2 night job so i'm not surprised i had not all the details.

However one thing may worth mentioning.
I have a BLTouch installed.
It is hot wired on the hotend to the Z-MAX which is D2, so i had to edit
the pin of Z_PROBE_PIN.
It resulted in G28 did not deployed the touch, so i had to add a condition
in Conditionals_LCD.h

505: #define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 &&
(ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) || *defined(Z_MIN_PROBE_PIN)*))

Thanks,
István

davidramiro, 2019-04-05:
Interesting, though I have experience with BLtouch. You might want to have a look at https://github.com/MNieddu91/Marlin-AI3M-2.0.x-BLTouch, it's a Marlin 2 fork that supports BLtouch. 

@MNieddu91 maintains it, as far as I know he had his BLtouch already working with this version.

I'll close this for now but if you come up with any other questions regarding Marlin 2, don't hesitate to ask.

closed 2019-04-05 by davidramiro

iherbak, 2019-04-05:
Thanks for your support! Have a great day!

David Ramiro <notifications@github.com> (időpont: 2019. ápr. 5., P, 17:11)
ezt írta: