davidramiro/Marlin-Ai3M #33: [FR] Mega-S Extruder maybe add values to the describtion

sello89, 2019-04-10:
Hey

i have the Mega-S Exruder mounted to the printer and tried a few prints with different seetings.
You allready write in your describtion to adjust the Steps per Unit.
I would add two more important changes for the extruder. Just for a start point of fine tuning.
The first is the feedrate for the extruder this works perfect for my printer at E25.00.
The second is the Acceleration für the Retract this should be 1500 not 3000.

Maximum feedrates (units/s):
M203 X500.00 Y500.00 Z6.00 E25.00

Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1500.00 R1500.00 T3000.00

Awesome work elsewise for this fantastic work on this special firmware!!!!

Marcel

veloc1ty, 2019-04-10:
Hi Marcel,

just a question out of interest, are these values for TMC2208 or stock stepper drivers?

Best regards,
Alex

sello89, 2019-04-10:
Hey Alex. I use the TMC2208. 
It works fine with or without linear advanced.

Marcel

veloc1ty, 2019-04-10:
Perfect, thanks. I have installed the TMC2208 last week and I'm still trying to find THE perfect settings. These look quite promising - Thanks!

Alex

veloc1ty, 2019-04-11:
Okay, after doing a couple of prints, I can confirm these settings work very well for the Mega-S with TMC2208 v1.1 and linear advance either enabled or disabled.

Thanks for sharing these, would be great to have them as default or in the Wiki as part of the instructions section.

Alex

davidramiro, 2019-04-14:
Thanks for the suggestions and testing, guys! I will definitely add them to the description soon, though a hard cap of 25mm/s on the feedrate setting is too drastic imo. The retract speed is something you should limit via slicer, not via feedrate. Other than that, I completely agree.

Interestingly, when I used the Mega-S firmware to read its default values, they were exactly the same as on the i3 Mega:
```
Maximum feedrates (mm/s):
M203 X500.00 Y500.00 Z6.00 E60.00
Maximum Acceleration (mm/s2):
M201 X3000 Y2000 Z60 E10000
Accelerations: P=printing, R=retract and T=travel
M204 P3000.00 R3000.00 T3000.00
```

There has been an update to the Mega-S firmware since, maybe they reduced the values already, would make sense. I don't have the time right now, but I think I'll look into the new Mega-S firmware within the next week and make a decision then.

sello89, 2019-04-14:
I reduced the feedrate of the extruder because the gear extruder don't do that high settings. I hope i guess right. At E60.00 the extruder loose sometimes steps. 

The feedrate is the rate for the extruding speed not for the retract ! Hope that is right. 
The retract speed i set to 30mm/sec for PLA.
If i use 45 i get errors in the prints. 

And the Retract Acceleration is not the retract speed. 

The extruder have a 3:1 gear. Thats the reason why 
i reduce the values. 

But i'm very happy about any other opinion!

Have a nice sunday guys.

davidramiro, 2019-04-14:
>The feedrate is the rate for the extruding speed not for the retract ! Hope that is right.

No, it's not, feedrate is just a general term for movement speed in Marlin. Doesn't have anything to do with feeding filament. 

Think of it like this - when printing regularly, the extruder will never feed at 25mm/s, let alone 60mm/s. That would be an obscene amount of plastic for a 0.4mm nozzle. The only time where it hits those speeds is at retraction and unretraction. By setting feedrate to 25mm/s and your retraction speed to 30mm/s, you are actually having retractions of 25mm/s. Feedrate is a hard movement limit.

>The retract speed i set to 30mm/sec for PLA.
>If i use 45 i get errors in the prints.

Which is exactly my point. To combat the issues, the general feedrate is not crucial here, retraction and unretraction speeds are. By limiting the feedrate, you have just also limited retraction speed. 

So basically, you are correct about suggesting lower retraction speed (and acceleration - don't worry, I know the difference 🙂) on the Mega-S extruder. And I agree. I just don't agree about introducing a hard speed limit in the firmware.

Hope that brings my point across better.
Have a great sunday yourself!

sello89, 2019-04-14:
Ok thanks for the explanation! I think that is the way for getting to understand every thing better in our all hobby the 3D printing.

So i reset the marlin feedrate to the standard 60.
And stay with value in the slicer for around 30 for the retract Speed.

davidramiro, 2019-04-14:
Don't get me wrong - you can limit the feedrate just as you like! No problem in doing that, it's the same result als limiting it in your slicer settings.

I was just talking about making changes to the base firmware itself, hence I was hesitant about a hard limit there. 

Anyways, tomorrow I will add a new section on that to the description. Thanks a lot for your tests. :)

commit a9dc6753 (2019-04-20): Add Mega-S extruder settings to readme

Close #33
closed 2019-04-20 by davidramiro