davidramiro/Marlin-Ai3M #34: [FR] No retraction / filament purge for M125

rohrbrecher, 2019-04-11:
**Describe your suggestion.**
When sending an M125 command (pause print, basically an M600 without exchanging the filament, great for putting in nuts, magnets, etc. into prints) the printer still retracts a lot and purges after sending the continue command. Because of the retract I often get clogs and the purge wastes a lot of filament.
It would be great if only M600 retracts and exchanges the filament but M125 should only retract a little bit and "park" the head.

**Reasoning**
It's a great feature for many things. You can either put in mechanical parts into prints, like nuts, magnets, keyrings, ... but it is also great if there is something bad going on the printbed (e.g. failed support structure) that needs to be cleaned before continuing the print.

Reference: http://marlinfw.org/docs/gcode/M125.html

davidramiro, 2019-04-14:
Valid suggestion, but as I see it, we currently do have exactly this behaviour in the firmware. At least for pausing the print via display, I used `M125 L2` to park and retract 2mm as you can see [here](https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/AnycubicTFT.cpp#L681). As for the purge length, that should default to 2mm as well, as seen [here](https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/Configuration_adv.h#L1002).

So maybe just using the `L` parameter to define a smaller retraction would do the trick?

Also - which command are you using to resume the print? I could add a parameter to that command to skip the purge if it's causing issues.

rohrbrecher, 2019-04-26:
Sorry for the delay, the mail somehow ended up in my spam.

To continue I use M108, same as with M600. I will try putting M125 L2 into the GCode to see if this works. 

By the way: Is it possible to increase the time before the hotend temp starts dropping?

davidramiro, 2019-05-01:
Honestly, I won't look into the `M108` command as it is handled by the `EMERGENCY_PARSER`, it's a crucial part of Marlin that handles urgent commands right before they enter the buffer for security reasons.

Depending on if you print via SD or USB, I'd recommend you to look into the `M24`/`M25`, `G27` and `M125` [GCode documentation](http://marlinfw.org/meta/gcode/), I believe your desired behaviour is possible with some added parameters.

On top of that, you can easily edit the default pause parameters yourself in [this section](https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/Configuration_adv.h#L981) of `Configuration_adv.h`. Right there, on line 1012 you will also find `PAUSE_PARK_NOZZLE_TIMEOUT`, which is the nozzle heating timeout on pause.

davidramiro, 2019-05-17:
@rohrbrecher Any update? Was it possible to achieve your desired behaviour via GCode?

rohrbrecher, 2019-05-25:
Yes, thank you. I changed the parameter in FW for more time to swap the filament and the filament purge also works with the added L parameter.

davidramiro, 2019-05-25:
Glad to hear that!

closed 2019-05-25 by davidramiro