davidramiro/Marlin-Ai3M #37: [BUG]

jeffgram, 2019-04-19:
Hello,
AnyCubic i3 Mega(Upgraded to Mega S).
My bed leveling cannot be fixed and the nozzle is very high after i have exactly followed your instruction. I guess adjusting manually with the paper for i follow all this instruction.   
After I flashed the Marlin 1.1.9(stock) FW, i have done all these: 
M502 
M500.
Mine is upgraded, so M92 E384 , M500 for my Mega S Titan extruder: Done
Calibrating extruder: adjusted for more than 2 times:Done 
Manual Mesh Bed Leveling: Done, Saved through EEPROM, then Reboot.
Mesh Leveling Print:
G28
G26 C H200 P5 R25 Q4.2 Z4 (nozzle touches well with filament purge, slightly oozing and quite thin printed on the bed but looks quite good but still messy after print)
Finally,  i restarted. When i wanted to print something else or anything else, the nozzle is too high. I really don't understand this.  Do i have to replace my hotend and extruder with the new ones?
It took few days..... day and night... haven't slept until right now you are seeing me with my red eyeballs open barely. 
Waiting for your response.

veloc1ty, 2019-04-19:
Hi Jeff, have you edited your slicer settings as well so it will use the mesh leveling? 
You need to set
M501
M420 S1
So the slicer will instruct the printer to actually use the mesh from the eeprom.

davidramiro, 2019-04-19:
Please attach your slicer's start Gcode.

jeffgram, 2019-04-19:
Thanks #Veloc1ty and Davidramiro.
I have done these Gcodes  'M501 & M420 S1' through 'Pronterface' Software after the Mesh Leveling, saved through, then rebooted. 
There was also a knot loose on my Mega S titan extruder, which I have just fixed it some while ago. Next, I did the extruder caliberation test and 25mm out of 120mm filament was left out, which I did the calculation and came out E404.21. Next I saved 'M92 X80.00 Y80.00 Z400.00 E404.21"with M500. Then, I did the Mesh Print Test gcode, which is  'G26 C H200 P5 R25 Q4.2 Z4' . The result came better than before but if i don't get myself wrong, the lines are thicker than before, yet looks better(I hope so as before one was a little bit thinner). I rebooted.
Finally, Whenever I printed any file, the nozzle was high enough and couldn't reach the bed. It was filling out in air, which made me confused.

Here is the Start Gcode:
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F{speed_travel} ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F{speed_travel}
M117 Printing...
G5
--------------------------------------------------
End Gcode:

M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M107
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle
to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops
so the head is out of the way
G1 Y180 F2000
M84 ;steppers off
G90
M300 P300 S4000

[image]

davidramiro, 2019-04-19:
`M501` and `M420 S1` need to be in your slicer start code. It needs to get called before every print. So, just like mentioned in the instructions, your start Gcode needs to look like this:

```
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
M501
M420 S1
G1 Z15.0 F{speed_travel} ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F{speed_travel}
M117 Printing...
G5
```

So your Mesh is fine and your printer is ready to use it, you just never enabled it on your prints.

closed 2019-04-19 by davidramiro

jeffgram, 2019-04-19:
After that, do i have to follow this one too?

"Manual commands for use with OctoPrint... etc. :
G29 S1 - Start MMBL
G29 S2 - Next Mesh point
Raising Z: G91 , G1 Z0.02 , G90 (one after another, not in one line)
Lowering Z: G91 , G1 Z-0.02 , G90
All 25 points done: After seeing ok in the console, send M500 to save."

veloc1ty, 2019-04-19:
Nope, just the start code in the slicer - that will instruct the printer firmware to use the mesh and that should be it.

jeffgram, 2019-04-19:
It worked finally!!!... thank you very much Voloc1ty!! as well as David!!