davidramiro/Marlin-Ai3M #38: [HOWTO] leave hotend with PID and set bed to bang-bang mode?

1f7, 2019-04-22:
Hi,
I want install 220V Silicone heating pad with solid state relay.
In this connection I want setup bed to bang-bang mode.
It is enough will be to comment out `PIDTEMPBED` in `Configuration.h` or need some more?

davidramiro, 2019-04-22:
Yes, simply commenting out `PIDTEMPBED` will enable bang-bang.

For further adjustments, you can take a loot at the following settings in `Configuration_adv.h` - those are the default settings from Anycubic - maybe 5000ms is too coarse for mains voltage:
```
#if DISABLED(PIDTEMPBED)
  #define BED_CHECK_INTERVAL 5000 // ms between checks in bang-bang control
  #if ENABLED(BED_LIMIT_SWITCHING)
    #define BED_HYSTERESIS 2 // Only disable heating if T>target+BED_HYSTERESIS and enable heating if T>target-BED_HYSTERESIS
  #endif
#endif
```

1f7, 2019-04-23:
Hello David!
I think the bed temp is inert and is not critical as on the hotend. 5 sec is normal.
I will check when heater is arrive
Thanks for the quick response!

closed 2019-05-01 by davidramiro

1f7, 2019-06-14:
I only comment `PIDTEMPBED`
All works fine, thank you David!