davidramiro/Marlin-Ai3M #39: [BUG] M600 Extruder to fast! (titan)

Armageddit, 2019-05-01:
**Describe the bug**
if ich use M600 for filament change ir runs very quickly 3 times and after this it runs with normal speed.
i have set a max speed with M92 E388.27 but this will be ignored :(

**To Reproduce**
Steps to reproduce the behavior:

use M600 :)

**Printer (please complete the following information):**
 - Model [i3 Mega with Titan extruder]

**M503 Output**

Printer is now online.
echo:V1.1.3
 1.1.0-RC8
echo: Last Updated: 2016-12-06 12:00 | Author: (Jolly, xxxxxxxx.CO.)
Compiled: Oct 24 2018
echo: Free Memory: 3109  PlannerBufferBytes: 1168
echo:V27 stored settings retrieved (398 bytes)
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E388.27
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z6.00 E60.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y2000 Z60 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Z2 Endstop adjustment (mm):
echo:  M666 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P9.02 I0.31 D42.76
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:SD card ok
echo:SD card ok

davidramiro, 2019-05-01:
```
echo:V1.1.3
1.1.0-RC8
echo: Last Updated: 2016-12-06 12:00 | Author: (Jolly, xxxxxxxx.CO.)
```

Uhm, you are not running this firmware...

Armageddit, 2019-05-03:
oh sorry :(

> Connecting...
> start
> Printer is now online.
> echo:Marlin 1.1.9
> Marlin-AI3M v1.4.6
> echo: Last Updated: 2019-04-03 | Author: (davidramiro)
> echo:Compiled: Apr  4 2019
> echo: Free Memory: 1362  PlannerBufferBytes: 1664
> echo:V55 stored settings retrieved (731 bytes; crc 3016)
> echo:  G21    ; (mm)
> echo:Filament settings: Disabled
> echo:  M200 D1.75
> echo:  M200 D0
> echo:Steps per unit:
> echo:  M92 X80.00 Y80.00 Z400.00 E388.27
> echo:Maximum feedrates (units/s):
> echo:  M203 X500.00 Y500.00 Z6.00 E45.00
> echo:Maximum Acceleration (units/s2):
> echo:  M201 X3000 Y2000 Z60 E10000
> echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
> echo:  M204 P1500.00 R3000.00 T3000.00
> echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
> echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
> echo:Home offset:
> echo:  M206 X0.00 Y0.00 Z0.00
> echo:Mesh Bed Leveling:
> echo:  M420 S0 Z0.00
> echo:Endstop adjustment:
> echo:  M666 Z0.00
> echo:PID settings:
> echo:  M301 P16.49 I1.03 D65.86
> echo:  M304 P243.91 I46.53 D319.67
> echo:Linear Advance:
> echo:  M900 K0.00
> echo:Filament load/unload lengths:
> echo:  M603 L538.00 U555.00
> echo:SD card ok
> echo:SD card ok

davidramiro, 2019-05-03:
I don't really understand your issue. The extruder is "running too fast"? Can you elaborate? And what do you mean by 3 times?

Your bug report is kinda sparse, just "use M600" is not enough information as I cannot reproduce this issue at all.

Also M92 sets steps per mm, has nothing to do with speed.

Armageddit, 2019-05-03:
> Also M92 sets steps per mm, has nothing to do with speed.
```
echo:Maximum feedrates (units/s):
echo: M203 X500.00 Y500.00 Z6.00 E45.00
```
sry my mistake :/ 
i think M92 is usen to calculate M203
all retracts with retract speed 45mm/s will work

> I don't really understand your issue. The extruder is "running too fast"? Can you elaborate? And what do you mean by 3 times?

it rotates very fast i cant see the direction because the wheel is standing still ant the stepper motor scratched. not enaugh tourque to turn the extruder so quick

davidramiro, 2019-05-03:
Ah, I see. You can change the advanced pause retract feedrate here:

https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/Configuration_adv.h#L983

Armageddit, 2019-05-06:
can this set to max of M203 or only in this Configuration bevor upload?
or can i change it with an M command?