davidramiro/Marlin-Ai3M #43: case statement 28 in AnycubicTFT.cpp appears incorrect

robnrie, 2019-06-01:
In line 1231 case 28 is incomplete.
          case 28: // A28 filament test
            {
              if(CodeSeen('O'));
              else if(CodeSeen('C'));
            }
            ANYCUBIC_SERIAL_ENTER();
            break;

davidramiro, 2019-06-02:
Indeed it does not do anything, this code is from [Anycubic's source](https://github.com/ANYCUBIC-3D/I3-MEGA/blob/master/Marlin_TFT_CR_V1.1.1/Marlin_main.cpp#L1413), not quite sure what they planned to do there. Since filament checking happens in another standalone function now, I will remove those few lines on the next commit.

robnrie, 2019-06-02:
Ok thanks just thought i would check as there source was pretty messed up and i wasn't sure if it was planned to do something with it later.

closed 2019-06-02 by davidramiro