davidramiro/Marlin-Ai3M #44: What changes needed for resume after power failure?

searchingforcode, 2019-06-01:
Hi is there any work going on for resume after power failure function? I can help also. Can you give me some hints?

davidramiro, 2019-06-02:
Hi there, 

to port it the way the manufacturer intended, you can have a look at their source code. Basically they do three things:

Modifying G5 GCode to initialize power loss resume: https://github.com/ANYCUBIC-3D/I3-MEGA/blob/master/Marlin_TFT_CR_V1.1.1/Marlin_main.cpp#L4163

Implement a function to save current position, stop motors and heaters: https://github.com/ANYCUBIC-3D/I3-MEGA/blob/master/Marlin_TFT_CR_V1.1.1/Marlin_main.cpp#L849

And finally a function to clear current values held in EEPROM and write the new position data: https://github.com/ANYCUBIC-3D/I3-MEGA/blob/305b0a041e99b0c964d1f7b20a15ae45da92718e/Marlin_TFT_CR_V1.1.1/configuration_store.cpp#L1052

Baking it straight into Marlin_main.cpp and using GCode as initialization is not the most beautiful way to do it imo, but it works. There are other ways to implement it, e.g. using the Creality power loss feature that got introduced to Marlin officially a while back.

Since I do not experience power outages, this feature is not really high up on my list of priorities but please let me know if you decide to have a go at porting it.

davidramiro, 2019-06-19:
Closing due to lack of response. Feel free to get in touch again later on.

closed 2019-06-19 by davidramiro