davidramiro/Marlin-Ai3M #45: [BUG] Touchscreen freezes on filament change pause.

AmyTheCute, 2019-06-07:
The filament change pause from the special menu causes the LCD screen to freeze. but the printer works fine and I can send the M108 command from octoprint to make it work.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print (from octoprint in my case)
2. Use Filament change pause from the special menu after a few layers
3. The touchscreen freezes. no *beep* sounds when you select an item anymore either. it's just fully frozen

**Expected behavior**
The touchscreen would continue functioning and lets me touch resume.

**Photos of the print / output of the terminal**
If applicable, add photos or console outputs to help explain your problem.

**Printer (please complete the following information):**
 - Firmware version: 1.1.9
 - Printing method: Octoprint
 - Model: Mega-S

**M503 Output**
Please paste the complete output of the `M503` command here.
Send: M503
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E404.21
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z12.00 E120.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1500 Y1500 Z500 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1500.00 R1500.00 T2000.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.20 E2.50
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z10.00
Recv: echo:  G29 W I0 J0 Z0.05750
Recv: echo:  G29 W I1 J0 Z-0.09500
Recv: echo:  G29 W I2 J0 Z-0.02750
Recv: echo:  G29 W I3 J0 Z0.13750
Recv: echo:  G29 W I4 J0 Z0.30750
Recv: echo:  G29 W I0 J1 Z0.21750
Recv: echo:  G29 W I1 J1 Z0.10500
Recv: echo:  G29 W I2 J1 Z0.11000
Recv: echo:  G29 W I3 J1 Z0.21000
Recv: echo:  G29 W I4 J1 Z0.27750
Recv: echo:  G29 W I0 J2 Z0.36250
Recv: echo:  G29 W I1 J2 Z0.15750
Recv: echo:  G29 W I2 J2 Z0.10250
Recv: echo:  G29 W I3 J2 Z0.12000
Recv: echo:  G29 W I4 J2 Z0.19250
Recv: echo:  G29 W I0 J3 Z0.40750
Recv: echo:  G29 W I1 J3 Z0.23750
Recv: echo:  G29 W I2 J3 Z0.15000
Recv: echo:  G29 W I3 J3 Z0.10250
Recv: echo:  G29 W I4 J3 Z0.11500
Recv: echo:  G29 W I0 J4 Z0.44500
Recv: echo:  G29 W I1 J4 Z0.26750
Recv: echo:  G29 W I2 J4 Z0.14500
Recv: echo:  G29 W I3 J4 Z0.05250
Recv: echo:  G29 W I4 J4 Z0.03750
Recv: echo:Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P14.98 I0.85 D66.26
Recv: echo:  M304 P251.78 I49.57 D319.73
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-0.80
Recv: echo:Stepper driver current:
Recv: echo:  M906 X815 Y820
Recv:   M906 I1 Z800
Recv:   M906 T0 E895
Recv: 
Recv: echo:Hybrid Threshold:
Recv: echo:  M913 X210 Y210
Recv:   M913 I1 Z4
Recv:   M913 T0 E35
Recv: 
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.45
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L538.00 U555.00

davidramiro, 2019-06-08:
Hi,

I tested several ways of using the filament change pause and none resulted in any freezes so far.

>The touchscreen freezes. no beep sounds when you select an item anymore either. it's just fully frozen

Might have been some weird kind of glitch, since the display itself is completely independent from the firmware. Did this occur more than once?
The beeping for example is baked into the hardware of the screen. If the firmware would halt, the screen would still provide its acoustic feedback. Hence I can't really fathom what happened on your machine.

If this is reproducable every time at exactly this point, I will definitely try and figure out what's going on.

AmyTheCute, 2019-06-08:
Yes. Everytime, No expect ions. Tho i was thinking. Could it be caused by any if my UART pins for my TMC22208? I do use the pins 14 20 21

davidramiro, 2019-06-10:
>Could it be caused by any if my UART pins for my TMC22208? I do use the pins 14 20 21

That might be, as I just tried it myself and it worked flawlessly.. Hard to say how UART would interfere with this feature, although pin 14 does get routed through the TFT daughterboard as can be seen here:

[image]

You could try enabling `ANYCUBIC_TFT_DEBUG` in `Configuration.h`, I included some debugging messages within the filament change feature that you could read out via the OctoPrint terminal.
But I am pretty sure this is not related to Marlin itself as the screen, even when not connected to anything but power, will retain functionality, as in you can still navigate the menus and it will beep on touch input. So a freeze like you are experiencing is rather something on the hardware side.

davidramiro, 2019-06-24:
Any further observations, @AmyTheCute?

I know it's probably too much to ask whether you could try this again without using UART jumpers (maybe it's really just pin 14?), but I am fairly sure this is not reproducible on a stock setup.

davidramiro, 2019-07-17:
Confirmed to not be reproducible on 2 different machines. I'm closing this issue due to lack of a response, but please do report back if you make any further observations.

closed 2019-07-17 by davidramiro

AmyTheCute, 2019-09-25:
Dirty for the lack of response i just saw this. I sadly no longer have the anycubic but i do confirm it was on my end. I accidentally messed up my pins.h file so, sorry!