davidramiro/Marlin-Ai3M #46: [BUG] Part cooling fan turns off if you change print speed from menu

b-desconocido, 2019-07-17:
**Describe the bug**
If you adjust print speed via printer's menu while still printing, the part cooling fan is turned off (or paused) until it spots M106 command in your file or via the serial terminal. I'm not sure if anything else is affected (it might be), but turning the part cooling fan is enough to ruin PLA-model.

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
2. Start printing
3. Wait until part cooling fan turns on, in my case at the second layer
4. Adjust the print speed to any value via menu and hit 'OK'
5. Enjoy the silence - part cooling fan is turned off
Or:
1. Open Repetier Host
2. Turn on part cooling fan and home the nozzle
3. Adjust the print speed to any value via menu and hit 'OK'
4. Enjoy the silence - part cooling fan is turned off

**Expected behavior**
The part cooling fan should be running? Or i'm wrong?

**Printer (please complete the following information):**
 - Firmware version: downloaded from 'Releases' a week ago, tmc2208 version
 - Printing method: SD, Serial
 - Model: Mega-S

davidramiro, 2019-07-17:
That's just how the menu works - there are two settings, fan speed and print speed. Both get set as you click on OK. Since the fan speed setting defaults to zero, that's why you see this behaviour.

Hence you would have to input the current fan speed too whenever you want to change the print speed so it does not get set to 0.

b-desconocido, 2019-07-17:
So the screen didn't fetch current value of fan? I didn't noticed that

davidramiro, 2019-07-17:
It does fetch it and shows it under "Current". But the first time you enter the Speed menu, the "Fan Speed" setting is always on 0. After you set it manually, it will remember it.

That's just one of the quirks of the screen, nothing I can do about it sadly since its firmware is closed source.

b-desconocido, 2019-07-17:
I get it, thank you! Do you know what MCU is used for the screen?

davidramiro, 2019-07-17:
Yes, it's a Nuvoton NUC120LC1BN. (Cortex M0)

Are you interested in reverse engineering it? I already tried doing so on the binaries that you can find on its USB interface to no avail, but that just isn't really my area of expertise 🙂

b-desconocido, 2019-07-17:
Yes, although i haven't much experience in embedded world. There was only a single attempt in obtaining firmware from LiPo charger based on 8051 MCU, but it had read out protection and i was't able to get original firmware. Debugging was working fine, but i couldn't get the data whatever i tried - software 'tricks' or physically glitching the Vcc voltage while it loading the option bytes (i even desoldered the caps).

Gonna read the datasheet, could you please share a files you extracted from USB? Are those are 'resources' or there is an actual firmware?

davidramiro, 2019-07-17:
Here you go: [screen binaries.zip](https://github.com/davidramiro/Marlin-Ai3M/files/3401601/screen.binaries.zip)

I doubt those are firmware binaries but I am curious to know what you can find out. I didn't get anything usable using Ghidra.

b-desconocido, 2019-07-17:
Yea, these are resources - font, images and some junk, probably menu descriptors. Need to get or take a closeup photos of screen module, there could be an ISP interface to program/debug MCU. So far I have no access to my machine.

b-desconocido, 2019-07-17:
Found a 'closeup' picture on Youtube: https://youtu.be/pqHJgDQ-qHs?t=69
There is definitely an ISP port (J4), i hope it is SWD-compatible. Other components looks pretty obvious - LDO regulator, backlight driver, MCU and 8-pin flash (?) chip. Although, I have a single picture of mine, and the MCU labeled as 'PenTableT blah blah numbers', but the board layout looks same, so it might be relabeled.

b-desconocido, 2019-07-17:
Hurray, according to datasheet:

```
– Support 2 wire ICP update through SWD/ICE interface
– Support fast parallel programming mode by external programmer
```
Application flash (APROM): 32KB
RAM: 4KB
Data flash (kind of EEPROM): 4KB
ISP ROM (for bootloader): 4KB

4KB of ram isn't much, but enough to render complex dynamic graphics using tiled rendering. Also i need to get the part number of LCD panel and touch IC, hope Anycubic could answer these questions, cuz reversing this isn't very funny.

I asked Anycubic to provide a spare screen, whatever they ask for it or send it for free, as i want to use printer meanwhile.

b-desconocido, 2019-07-17:
So, i found the ancestor of display module. Most likely it produced by Chinese company Easttop Display Co. Ltd and customized for Anycubic. It is either based on module M7-M350A1 or M7-H350A1. I'm trying to contact them to get the datasheets :)
Most likely no one would response, so i'm continuing my own "investigation". ISP port has the following pinout:
[image]
There is no boost circuit for back light, I just confused the buzzer with an inductor :) Despite that, it has 3.3V LDO, 32Mbit flash memory, and some relabeled MCU, which looks like the one you mentioned.
In case it is NUC120LC1BN, there might be a chance it is possible to extract firmware. According to datasheet, you could get it using ISP debugger despite the state of protection bit in Config0. I will try it soon.

b-desconocido, 2019-07-18:
I got a firmware :)
[tft firmware.zip](https://github.com/davidramiro/Marlin-Ai3M/files/3406399/tft.firmware.zip)

b-desconocido, 2019-07-20:
I made a tool to modify images on TFT: https://github.com/b-desconocido/Anycubic-Mega-TFT-Resource-Converter
Gonna try to implement open-source firmware for MCU soon

davidramiro, 2019-07-20:
Wow, great job! Please keep me updated!

b-desconocido, 2019-07-21:
Not much else we can do with files accessible through USB. You can change images, font and the loading screen animation. All the rest is hardcoded in firmware. It isn't that tough to reverse engineer, but it is almost impossible to add features to existing firmware or modify it in any way due to way the code is written and free space left. More feasible solution is to write a custom firmware from scratch, although i have no clues how it should look like, what options or settings must be in, nor how it should communicate with host MCU via serial, e.g. which commands, responses and arguments must be implemented.

davidramiro, 2019-08-27:
Sorry for the late reply and thank you for your efforts! This was a great insight, I never would have the guts to mess with the screen itself.

In theory the screen firmware should be really simple. It just forwards inputs, which get processed by AnycubicTFT.cpp and then executed by calling native Marlin functions. Anyways, if you wanted to get more functionality out of the screen, I would recommend replacing it with some Marlin compatible 12864 screen. Then you get all the native menus and lots of functions you can enable which only work with a native LCD.

Thanks again, I'm closing this issue for now. But please keep me updated if you gain any more insights!

closed 2019-08-27 by davidramiro