davidramiro/Marlin-Ai3M #47: [BUG] One motor doesnt stop when homig Z

vvranjek, 2019-08-10:
**Describe the bug**
When homing Z axis, one motor (the right one, on the side of the end-stop button) doesn't stop and keeps grinding until I turn off the printer.

Im also a dev, I can answer technical questions, but I am not accustomed with the code much. You can point me to code where I can debug myself. Thanks! 

**To Reproduce**
- Flash  firmware 1.4.6-stock-drivers (also tried 1.3.4 and 1.4.5 with same results) using Cura 4.2.1 
- Run M502 and M500
- Run G28 Z (Home Z)
- Watch the printer destroy itself

**Expected behavior**
All motors should stop when hitting the end-stop button

**Printer (please complete the following information):**
Anycubic i3 Mega, I bought it very soon after the release

**M503 Output**

[17:13:05:109] echo:  G21    ; (mm)␊
[17:13:05:110] ␊
[17:13:05:110] echo:Filament settings: Disabled␊
[17:13:05:111] echo:  M200 D1.75␊
[17:13:05:114] echo:  M200 D0␊
[17:13:05:114] echo:Steps per unit:␊
[17:13:05:114] echo:  M92 X80.00 Y80.00 Z400.00 E92.60␊
[17:13:05:116] echo:Maximum feedrates (units/s):␊
[17:13:05:117] echo:  M203 X500.00 Y120.00 Z6.00 E60.00␊
[17:13:05:118] echo:Maximum Acceleration (units/s2):␊
[17:13:05:120] echo:  M201 X2000 Y1200 Z60 E10000␊
[17:13:05:121] echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>␊
[17:13:05:124] echo:  M204 P1200.00 R3000.00 T1500.00␊
[17:13:05:126] echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>␊
[17:13:05:131] echo:  M205 Q50000 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00␊
[17:13:05:133] echo:Home offset:␊
[17:13:05:134] echo:  M206 X0.00 Y0.00 Z0.00␊
[17:13:05:136] echo:Mesh Bed Leveling:␊
[17:13:05:137] echo:  M420 S0 Z0.00␊
[17:13:05:137] echo:Endstop adjustment:␊
[17:13:05:139] echo:  M666 Z0.00␊
[17:13:05:139] echo:PID settings:␊
[17:13:05:141] echo:  M301 P15.94 I1.17 D54.19␊
[17:13:05:141] echo:  M304 P251.78 I49.57 D319.73␊
[17:13:05:143] echo:Linear Advance:␊
[17:13:05:143] echo:  M900 K0.00␊
[17:13:05:146] echo:Filament load/unload lengths:␊
[17:13:05:146] echo:  M603 L530.00 U555.00␊
[17:13:05:148] ok␊\

Cheers, and thanks for the great work!

davidramiro, 2019-08-10:
Thanks for the detailed report. In this case this is luckily not a bug. You said it yourself - you got a very early model, those only have Z-endstops on one side instead of both. 

To fix your issue, you will simply need to edit one line in the configuration and recompile: https://github.com/davidramiro/Marlin-AI3M/wiki/Customization-&-Compiling

In the linked guide there is a menu point about the old model. Let me know if this worked!

Please note that Anycubic doesn't distribute stock firmwares for single endstop machines anymore on their webpage. If you need to go back to the stock fw, you will have to contact their support, they will provide you with the single endstop fw.

vvranjek, 2019-08-11:
Thanks for a quick reply!

After some more fails, I managed to get it working. As so many times before, the trick was in thoroughly reading the instructions! :) The mesh bed leveling works great. Thanks for the help keep up the good work!

closed 2019-08-11 by vvranjek

Jamiecharris, 2019-09-24:
> Thanks for a quick reply!
> 
> After some more fails, I managed to get it working. As so many times before, the trick was in thoroughly reading the instructions! :) The mesh bed leveling works great. Thanks for the help keep up the good work!

I have the same issue, would it be possible to provide the hex file you compiled to see if it would work on my printer?

vvranjek, 2019-09-25:
Sure, here you go and good luck! (Unzip it first)

[Marlin-Ai3M-v1.4.6-custom.zip](https://github.com/davidramiro/Marlin-Ai3M/files/3651273/Marlin-Ai3M-v1.4.6-custom.zip)