davidramiro/Marlin-Ai3M #48: [BUG] Endstop BEEP on second Z axis

fedot01, 2019-09-26:
**Describe the bug**
There is no BEEP sound when endstop on right axis closed. But motor stopping successfully.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile and flash last firmware with `#define ENDSTOP_BEEP` at `Configuration.h`
2. Manualy close endstop on right Z axis
3. ???
4. NO BEEP!

**Expected behavior**
Should be BEEP as at other endstops.

**Printer (please complete the following information):**
 - realese v1.4.6
 - i3 Mega

schnello, 2019-10-01:
Hello.

This works as designed. (So create a ticket in the Marlin git)
`static char last_status=((READ(X_MIN_PIN)<<2)|(READ(Y_MIN_PIN)<<1)|READ(X_MAX_PIN));
`

From my personal point of view, the function does exactly what it should. A notification when the axis is at 0.

davidramiro, 2019-10-02:
This is intended, I figured that you only need one beep per axis. Especially since the second Z endstop usually gets triggered at the same time as the first one if the axis isn't heavily misaligned. It just saves a few CPU cycles just polling three sensors instead of four all the time.

But you can easily enable it, by replacing `(READ(X_MIN_PIN)<<2)|(READ(Y_MIN_PIN)<<1)|READ(X_MAX_PIN)` with `(READ(X_MIN_PIN)<<3)|(READ(Y_MIN_PIN)<<2)|(READ(Y_MAX_PIN)<<1)|READ(Z_MIN_PIN)` in the `EndstopBeep` function at line 14840 of [Marlin_main.cpp](https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/Marlin_main.cpp#L14840).

Keep in mind that you also probably need to decrease the current delay of 300ms between beeps (`static millis_t endstop_ms = millis() + 300UL;` a few lines further down) as I implemented that to avoid multiple beeps in rapid succession.

closed 2019-10-02 by davidramiro

fedot01, 2019-10-02:
Thanks for that code.
I wanted that beep on second Z endstop to adjust screw which hit it.

davidramiro, 2019-10-02:
Ah, I understand.

```
#ifdef ENDSTOP_BEEP
  void EndstopBeep() {
    static char last_status=((READ(X_MIN_PIN)<<3)|(READ(Y_MIN_PIN)<<2)|(READ(Y_MAX_PIN)<<1)|READ(Z_MIN_PIN));
    static unsigned char now_status;

    now_status=((READ(X_MIN_PIN)<<3)|(READ(Y_MIN_PIN)<<2)|(READ(Y_MAX_PIN)<<1)|READ(Z_MIN_PIN))&0xff;

    if(now_status<last_status) {
      static millis_t endstop_ms = millis() + 200UL;
      if (ELAPSED(millis(), endstop_ms)) {
        buzzer.tone(60, 2000);
      }
    last_status=now_status;
    } else if(now_status!=last_status) {
      last_status=now_status;
    }
  }
#endif
```

I think this should work well.