davidramiro/Marlin-Ai3M #49: Extruder Heatsink Fan is Controlled with Part Cooling Fan[BUG]

wufishy, 2019-10-02:
**Describe the bug**
Extruder heatsink fan speed is set to the speed of part cooling fan while running g-codes from octoprint or direct from SD card.  i.e. If I am not using part cooling fan the heatsink fan also turns off. I can manually set the part cooling fan speed via octoprint terminal (m106 S100) and separate control is restored.

**Expected behavior**
heatsink fan will run at full speed whenever the extruder is above 50 degrees per firmware setting.

**Printer (please complete the following information):**
 - Firmware version: Marlin-Ai3M-v1.4.6-TMC2208_reversed.hex 
 - Printing method: SD, Octoprint; sliced by cura
 - Model: Mega-S

**M503 Output**
Send: M503
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E500.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E30.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000 Y2000 Z60 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R1500.00 T500.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y-10.00 Z0.00
Recv: echo:Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P25.37 I1.67 D96.13
Recv: echo:  M304 P165.35 I17.40 D392.82
Recv: echo:Linear Advance:
Recv: echo:  M900 K10.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L538.00 U555.00
Recv: ok

Thanks in advance

davidramiro, 2019-10-02:
Weird, I can not reproduce this. My heatsink fan is, as you would expect, always running at full speed if over 50°C. Would you be able to send over a GCode file where this happens?

wufishy, 2019-10-03:
Yes of course, thanks
[AI3M_Bed_Level_Test.zip](https://github.com/davidramiro/Marlin-Ai3M/files/3683946/AI3M_Bed_Level_Test.zip)

davidramiro, 2019-10-08:
Thank you. Not quite sure what you are experiencing, but in my case:

1. Printer heats up, heatsink fan spins up.
2. Homing and priming happens
3. Parts cooling fan starts, heatsink fan stays on
4. Bed level test is printed
5. Parts cooling fan turns off, printer homes to X
6. Heatsink fan turns off when nozzle has cooled down

This would also be the expected behaviour, so I'm not sure what I am looking for here?

davidramiro, 2019-10-08:
Sure thing, here you go.

[Marlin-Ai3M-v1.4.6-TMC2208_reversed.hex.zip](https://github.com/davidramiro/Marlin-Ai3M/files/3703029/Marlin-Ai3M-v1.4.6-TMC2208_reversed.hex.zip)

But I suspect there might as well be some hardware glitch going on, as you're the first one to report that issue.

davidramiro, 2019-10-22:
Any further info on this?

wufishy, 2019-10-23:
I tested the firmware you've compiled in conjunction with the latest octoprint update and it's been fixed.

Thanks for your help :)

closed 2019-10-23 by wufishy