davidramiro/Marlin-Ai3M #50: filament change problem, printer stuck

davi-12, 2019-10-19:
Hi Davidramiro, first of all I would like to congratulate you for the great work done in this firmware, I solved the problem of the plate not perfectly plane But I have a problem with changing the filament, I use cura 4.2.1 and when I set pause to height, I save the file in the sd memory, I start printing from the sd memory, when the printer pauses, I change filament but at  the moment the sensor no longer detects the filament, the printer stops.  I tried to change the filament and press "continue" but it remains locked.  I can only turn it off and on to restart it.  do you have any advice for me? Can you explain how to enter the M600 command for sd card printing?  Do I have to insert it before the desired layer or immediately after? Thank you

davidramiro, 2019-10-19:
Hi there,

don't use Cura's pause plugin, just open the GCode file with a text editor, search for "layer X" or "layer: X" (depending on slicer) and insert M600 right before it. That way you can then change the filament and use the on screen "continue" button. Hope that helps!

davi-12, 2019-10-19:
Hi David,
Thank-you very much for your suggest, now the change of filament....works
very well!!! I am very pleased for your help 😊
Best wishes Davide

davidramiro, 2019-10-19:
Glad to hear that!

closed 2019-10-19 by davidramiro