davidramiro/Marlin-Ai3M #51: Trigorilla 1.1 Board causes display to loop on filament sensor warning

pixeliteUE, 2019-10-22:
**Problem I am experiencing**
When flashing the .hex of the latest version - the printer will fall into an endless loop of a lack of filament message. This stops commands from running on terminals, and commands such as M105 get sent but never hear a response because of this.

Flashing stock firmware via arduino IDE causes the machine to start chiming on loop instead.

I can't go back to the old trigorilla board due to it having issues with some connections, therefore I am stuck with the 1.1 board.

davidramiro, 2019-10-22:
>I can't go back to the old trigorilla board due to it having issues with some connections

Are you sure you have everything connected correctly on your new board? This firmware works fine on the newer 1.1 boards. A loop like that would occur when there are erraneous readouts from the sensor or if it's not connected.

>Flashing stock firmware via arduino IDE causes the machine to start chiming on loop instead.

Try flashing back via Cura. I have not tried flashing the stock fw via Arduino, although via Cura it went without an issue.

If you don't actually use the filament sensor, you can comment out `ANYCUBIC_FILAMENT_RUNOUT_SENSOR` in Configuration.h, so the polling of the sensor would be skipped.

pixeliteUE, 2019-10-22:
> Try flashing back via Cura. I have not tried flashing the stock fw via Arduino, although via Cura it went without an issue.

Doing this triggers the issue. However commenting out `ANYCUBIC_FILAMENT_RUNOUT_SENSOR `in the source has made this problem go away. Thanks! Now to deal with only 2 of the endstop sensors working..

davidramiro, 2019-10-22:
Isn't the filament sensor also connected at the Trigorilla's endstop pins? I'm sure there is just some minor issue or mixup on your wiring, if both the filament and the endstop sensor are having issues.

You can use M119 to debug/test endstop states, maybe that helps in narrowing down the issue. Good luck!

pixeliteUE, 2019-10-22:
> Isn't the filament sensor also connected at the Trigorilla's endstop pins?
It popped out of the socket months ago and I never bothered connecting it back in. The printer works fine without the filament sensor (before this) and after commenting it out that issue goes away. But the endstop issue remains.

Manually triggering z_min causes the usb connection to reset.
Manually triggering z2_min reports it as TRIGGERED in M119.
Manually triggering x_min reports it as TRIGGERED, however for some reason it tried to home on the right rather than the left.
Manually triggering y_min reports it as open and nothing happens, apart from when homing the buildplate rams itself into the rear

I've attached a photo of the wiring [https://photos.app.goo.gl/dbUjjLyxZeae6izc7](url)

davidramiro, 2019-10-22:
>however for some reason it tried to home on the right rather than the left.

You might just need to reverse the motor direction, since you compiled it yourself and by default, the config is set for TMC2208.

>Manually triggering z_min causes the usb connection to reset.

That is really weird. Sorry, I really have no idea what could be wrong here. The only further advice I have would be to enable `PINS_DEBUGGING` in Configuration_adv.h and use the [M43 Gcode](http://marlinfw.org/docs/gcode/M043.html), that way you can monitor which pins are actually being triggered.

pixeliteUE, 2019-10-22:
Here is the M43 output: https://pastebin.com/3fH1qJRh

What i've noticed is that Z2_MIN_PIN seems to be missing, any idea as to why that might be?

davidramiro, 2019-10-22:
Z2_MIN_PIN == X_MAX_PIN, which is 43 on your output and thus correctly defined.

Since, in fact, every pin is defined correctly in your firmware, I strongly suggest you take a closer look at the hardware side. Try and reconnect the filament sensor connector, maybe its common ground gets used for another endstop sensor.

pixeliteUE, 2019-10-22:
> maybe its common ground gets used for another endstop sensor
I don't think that would be the case, as the endstops worked fine without the sensor with the old board. And from what i've seen diagram wise Trigorilla 1.1 is basically identical to 1.0

davidramiro, 2019-10-22:
Well then it might not be exactly that. I don't have the exact PCB/wiring layout in memory. Can you please still double check your wiring? I just don't know how this would be related to my fork of Marlin specifically.

>Manually triggering z_min causes the usb connection to reset.
Manually triggering x_min reports it as TRIGGERED, however for some reason it tried to home on the right rather than the left.
Manually triggering y_min reports it as open and nothing happens, apart from when homing the buildplate rams itself into the rear

Since we can rule out a software issue - your pin config is correct, and as you pointed out yourself, pin layout is identical to 1.0 - this is quite obviously some hardware issue, hopefully just some mixup in wiring.

pixeliteUE, 2019-10-22:
I redid the wiring again, and seem to still experience the issue.

davidramiro, 2019-10-22:
I suppose you might have a defective board then, this firmware definitely works fine out of the box with rev. 1.1 🙁

closed 2019-10-22 by davidramiro

pixeliteUE, 2019-10-22:
Tried to see if it was a board issue by using the old board which worked just fine, and is experiencing the same issues with the endstops. Somethings wrong somewhere but not sure what