davidramiro/Marlin-Ai3M #52: [BUG] Hotend cools down during printing, PID tuning doesn't help

AlexTOOT, 2019-10-24:
**Describe the bug**
In preheat, hotend manages to heat up to 240°C and maintains stable. but when I start to print for a couple of minutes, it starts to cool down to 230°C and even lower. later the thermal protection seems to kick in and all motor freezes, the print just stops.
Notice : when it stops printing, the screen loses control, it cannot read the correct temperature. The pause/stop button can be clicked but it will just show please wait forever.
Tried M303 E0 S240 C15 U1 to tune hotend PID and nothing different happens. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
2. Preheat ABS
3. Throw any gcode with above 220°C via SD card
4. A few minutes later, It just stops printing

**Expected behavior**
Expect it to print normally without cooling down.
I know it's very likely that I did someting wrong instead of a bug. Could you please help me out?

**Printer (please complete the following information):**
 - Firmware version: V1.4.6
 - Printing method: SD, sliced with Cura
 - Model: i3 mega with single Z endstop

**M503 Output**
Please paste the complete output of the `M503` command here.

davidramiro, 2019-10-24:
Well, it definitely is hard for the i3 Mega to hold those kind of temperatures, but it should be possible.

How are you running your parts cooling fan? Do you use an enclosure for printing ABS?

>Notice : when it stops printing, the screen loses control, it cannot read the correct temperature. The pause/stop button can be clicked but it will just show please wait forever.

Nothing to worry about there, thermal runaway protection just freezes the board completely, the display is working as a standalone and cannot communicate anymore.

martblek, 2019-10-27:
Hi,
Try to lower cooling percentage to 75%.
My radial fan at 100% decrease temperature from 240 to 220 and lower then system stops.

davidramiro, 2019-11-15:
Closing due to lack of response. Feel free to reopen if you still have the issue!

closed 2019-11-15 by davidramiro

AlexTOOT, 2019-12-02:
> Well, it definitely is hard for the i3 Mega to hold those kind of temperatures, but it should be possible.
>
> How are you running your parts cooling fan? Do you use an enclosure for printing ABS? [...]

I tried turning off the partcooling fan completely, temp still drops which confuses me. 
As for closure.... I just shovelled the whole thing into my closet LOL.

Sorry for lack of response, had a few busy days and lost access to github. GFW in China, you know.