davidramiro/Marlin-Ai3M #55: [FR] Connections sound , filament empty sound

shaddow501, 2019-11-01:
Hello

Where in the firmware can I configure the USB connect sound (when the cura connect to the printer with the old firmware there was a sound) how do I configure it?

Also there is no sound when filament sensor do not detect filament...

davidramiro, 2019-11-04:
Uncomment line 1538 in Configuration.h.

```
//
// STARTUP CHIME
//
// Play a (non-earpiercing) startup chime on startup/serial connection
// of the Trigorilla board
//
//#define STARTUP_CHIME
```

I altered the sound, I found the old one to be way too high pitched 🙂

closed 2019-11-04 by davidramiro

shaddow501, 2019-11-04:
Thanks, do you maybe know how to add  lack of filament sound?

shaddow501, 2019-11-05:
Ok Solevd it

Took the  music.h & music.cpp file and integrated it into this firmware 
https://github.com/ANYCUBIC-3D/I3-MEGA/blob/305b0a041e99b0c964d1f7b20a15ae45da92718e/Marlin_TFT_CR_V1.1.1/music.h

https://github.com/ANYCUBIC-3D/I3-MEGA/blob/305b0a041e99b0c964d1f7b20a15ae45da92718e/Marlin_TFT_CR_V1.1.1/music.cpp

integrated he power up sound in: marlin_main.cpp

+#include "music.h"

At line 15393
 lcd_init();
  **PowerOnMusic();**
  lcd_reset_status();

With the filamentlack sound I have added in:  anycubicTFT.cpp
+ include "music.h"

and at line; 714 added the filament sound 

void AnycubicTFTClass::FilamentRunout()
{
  #if ENABLED(ANYCUBIC_FILAMENT_RUNOUT_SENSOR)
    FilamentTestStatus=READ(FIL_RUNOUT_PIN)&0xff;

    if(FilamentTestStatus>FilamentTestLastStatus) {
      // filament sensor pin changed, save current timestamp.
      const millis_t fil_ms = millis();
      static millis_t fil_delay;

      // since this is inside a loop, only set delay time once
      if (FilamentSetMillis) {
          **FilamentLack();**
        #ifdef ANYCUBIC_TFT_DEBUG
          SERIAL_ECHOLNPGM("DEBUG: Set filament trigger time");
        #endif
        // set the delayed timestamp to 3000ms later
        fil_delay = fil_ms + 3000UL;
        // this doesn't need to run until the filament is recovered again
        FilamentSetMillis=false;
      }