davidramiro/Marlin-Ai3M #58: [FR] Please enable filament sensor

Argurth, 2019-11-14:
**Describe your suggestion.**
The Anycubic Mega-S come with a filament sensor. It seems it is not activated on this firmware. Could it be possible to trigger a M600 when the filament sensor detect the end of filament, so that we can change the spool to another one ?

**Reasoning**
As it is a main feature of the Anycubic Mega-S, and it is supported by the stock firmware, we're loosing a feature using this firmware (event if overall, we win way more than we loose by using it instead of stock). I think others Anycubic Mega-S users would be pleased to be able to keep this feature, and as Anycubic is selling official parts, it would allow Anycubic Mega i3 users to easily upgrade if they wish to.

davidramiro, 2019-11-15:
I don't quite understand, the filament sensor works and does indeed park the head, doesn't it? I built the function for it right here:

https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/AnycubicTFT.cpp#L749
https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/AnycubicTFT.cpp#L246

If it does not work for you, I have included several debugging outputs, You can enable them by uncommenting `#define ANYCUBIC_TFT_DEBUG` on the bottom of Configuration.h and read the terminal output as soon as a filament runout happens.

Argurth, 2019-11-15:
Oh, ok, I thought it wasn't included because it stopped working since I installed the new firmware. I will try and debug it after my workday, thanks !

davidramiro, 2019-12-08:
Could you find out something? I could not reproduce the issue on my end. 

I do have an i3 Mega though, not a Mega-S. Maybe they changed out the sensor in later revisions, so I'd need to make some adjustments for that. If you could check with debugging, that would at least tell us if the firmware acknowledges the state of the sensor at all.

mssmison, 2019-12-08:
I can confirm that it's not working on the mega s. I'm sorry but I haven't
looked at the debugging yet though.

jdpurcell, 2019-12-13:
At least on my Mega-S (purchased February 2019), filament runout detection works fine with 1.4.6 firmware.

davidramiro, 2019-12-14:
Interesting, maybe Anycubic changed the filament sensor along the way? If you could provide some pictures that'd be helpful.

Arut42, 2019-12-14:
I have mine (i3 Mega S) runing on Marlin 2.0, Filament Sensor is working.
Maybe you have the wrong pin.
  #define FIL_RUNOUT_PIN    19
But I have problems with wrong trigger and am looking for a delay or filter.

davidramiro, 2019-12-14:
>But I have problems with wrong trigger and am looking for a delay or filter.

You can raise the delay here: https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/AnycubicTFT.cpp#L744

3000 ms works well for the regular i3 Mega, maybe you need some more time for the Mega-S sensor. Please report back!

guerradr, 2020-02-02:
Hi 
First of all thanks for the great firmware! I updated this week and I did have the same Issue that the filament runout sensor did not work. I think I found the Problem. I own one of the fist Series I3 Mega's and I have it in stock config. So I used your stock hex. file to flash. Apparentely the filament runout sensor is not activated in that particular compiled firmware. I compiled one myself and it worked flawlessly. 
Greetings

davidramiro, 2020-02-04:
>I think I found the Problem. I own one of the fist Series I3 Mega's and I have it in stock config. So I used your stock hex. file to flash. Apparentely the filament runout sensor is not activated in that particular compiled firmware. I compiled one myself and it worked flawlessly.

Oh, that could very well be. I'll look over this and recompile it properly tomorrow.

SFPROXY, 2020-02-14:
Hi. I have an Anycubic i3 Mega S and wanted to add that I'm also having this issue. The mainboard PCB has "Trigorilla 1.1" on it.

At first I thought maybe I had unplugged/plugged into the wrong spot, the filament sensor (I was also doing the TMC2208 swap on X & Y) but after reverting to "stock" _Marlin_TFT_MEGA_S_V1.1.5.ino_ from Anycubic, I confirmed that the filament sensor was in fact plugged in and working.

I enabled TFT debug and I'm getting the generic:

`TFT Serial Debug: Filament runout... J15`

I wish there was a better way to debug or get more information but considering it is a simple switch that's either on or off, I'm not sure.

I also increased the wait time from 3000 to 6000 with no better result. Interestingly the behaviour of the TFT menu is straight-up different from this Marlin, notably it says "USB Online" when I connect to it and it says nothing about filament in the spot that it does with custom Marlin. But, the beeps are all there (endstops, bootup, USB connected, etc) and the Pronterface read out confirms I am on stock firmware. Not surprising since I'm sure there's a lot of custom stuff and rewrites to get the TFT working at all.

I don't think there's much we can do at this point. I also tried setting the FIL RUNOUT PIN to 19 as one user of the Mega S states above and I saw on another forum, no go.

I know that the author here does not own a Mega S and sadly this is my uncle's which I am taking out of state to give back to him tomorrow so I will not be of any help in fixing this after tonight. I just wanted to confirm this is an issue for anyone wondering. It may vary from model to model even within the Mega S family. You know what the S must stand for right? haha...

trunneml, 2020-02-14:
Same Problem here. Filament sensor didn't work with this firmware. I've got an anycubic i3 mega with the mega-s upgrade kit installed.

davidramiro, 2020-02-14:
Hard for me to troubleshoot this since every single report of the filament sensor not working was on a Mega-S and the issue only started appearing a few months ago. I still suspect something got changed along the way, hardware-wise. Maybe it has something to do with the new Trigorilla revision v1.1?

Anyways, I have cleanly recompiled the latest build on an updated version of the Arduino IDE since @guerradr reported back that it was only buggy on the precompiled files. I have updated the latest v1.4.6 release tag, if you want to download it and try flashing that.

I'm kind of out of ideas. v1.4.6 was released in April 2019 and there are plenty of Mega-S machines that I know of where the filament runout works perfectly fine. So to troubleshoot this, we would need to find out what exactly has changed hardware-wise.

trunneml, 2020-02-14:
@davidramiro My i3 Mega is an old version where I only changed the extruder to the S-Version, so the board is the same. I will check the recompiled download.

trunneml, 2020-02-14:
@davidramiro Printer now shows the filament warning message in the display when the filament is running out. But octoprint doesn't stop printing. -.-
I think the new version is working and I'll have to check my octoprint setup.

Thanks

Michael

davidramiro, 2020-02-14:
> @davidramiro Printer now shows the filament warning message in the display when the filament is running out. But octoprint doesn't stop printing. -.-
> I think the new version is working and I'll have to check my octoprint setup.

That's good news, just one question, are you printing via Octoprint or via SD and just monitoring via Octoprint? Filament runout functionality only works on SD printing.

trunneml, 2020-02-14:
I'm currently printing via octoprint and yes via sd-card it should work. With the old firmware version no alert was shown on the display. Now it clearly says out of filament but octoprint don't recognizes it. So for me the problem is solved. Thanks for the firmware!

Argurth, 2020-02-17:
For people who want octoprint support, there is an octoprint plugin, I think it's Filament Sensor Revolutions, that can send M600 on a filament run-out detection.

As this firmware support M600, it work pretty well, that's what I've been using in the meanwhile, you will just have to unplug your sensor from the I3 and plug it to your Raspberry PI

davidramiro, 2020-02-17:
Thats good advice.

As for the initial bug report - I think it was just the precompiled hex files that were bugged as @guerradr found out, I have since replaced them and at least @trunneml reported back that it was indeed that issue. Guess we can consider it fixed?

Argurth, 2020-02-28:
I guess you can, I haven't the time needed to try out atm, but if the problem is still here when I do, I would reopen it

closed 2020-02-29 by davidramiro

sjscicluna83, 2020-07-17:
I’m also facing this issue but I only print with octoprint I know that my sensor is working as my ringer beeps when I do a filament change so as stated I think most users with this issue may be running octoprint.

My octoprint instance runs from a Linux machine not a raspberry pi so I can’t add a sensor to GPIO pins but I can do what’s mentioned in the below link, I haven’t tested it as yet but when I do I’ll report back here

https://community.octoprint.org/t/octoprint-windows-pause-octoprint-when-printer-sends-m600-command/13067/5

And a big thank you for a fantastic firmware