davidramiro/Marlin-Ai3M #59: [BUG] x home is set to x endpoint at the end of a print

Hoppaz, 2019-11-18:
I start the printer and release the motor and set the x position to the left side.
After this I start a print job (chessman) via octoprint 1.3.12 on octopie 0.16.0.
At the end of the job the hotend strays at the right hand side of the printed model but at the end of the plate. If I press X home (via lcd or octoprint) it will not move left. Only right and after this left to the point which seem to be the new X home. If I repeat it the next jobs are shifted to right the same way.

 - Firmware version: 0.19.0 ( "Anycubic i3 Mega / Mega-S Marlin 1.1.9 Custom Firmware - Extra Features & Quality Tweaks") from  https://github.com/davidramiro/Marlin-AI3M/releases -> Marlin-Ai3M-v1.4.6-stock_drivers.hex
 - Octoprint (USB)
 - i3 Mega

**M503 Output**
Send: M503
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E92.60
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E60.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X2000 Y1500 Z60 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P4000.00 R3000.00 T4000.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 Q20000 S0.00 T0.00 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P15.94 I1.17 D54.19
Recv: echo:  M304 P251.78 I49.57 D319.73
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L538.00 U555.00
Recv: ok

davidramiro, 2019-11-18:
Please attach the start and end gcode from your slicer configuration.

Hoppaz, 2019-11-18:
start code of the printed model:
;FLAVOR:Marlin
;TIME:1282
;Filament used: 1.06937m
;Layer height: 0.15
;MINX:8.841
;MINY:18.692
;MINZ:0.1
;MAXX:42.241
;MAXY:52.092
;MAXZ:19.9
;Generated with Cura_SteamEngine 4.3.0
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F50 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F50
M117 Printing...

---

end code of the printed model:

;TIME_ELAPSED:1282.738392
M140 S0
M204 S4000
M205 X20 Y20
M107
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M107
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F50 ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops, so the head is out of the way
G1 Y180 F2000
M84 ;steppers off
G90
M300 P300 S4000
M82 ;absolute extrusion mode
M104 S0
;End of Gcode

Hoppaz, 2019-11-18:
looks like it does not appear printing from the sdcard - still testing

davidramiro, 2019-11-18:
Probably has to do with OctoPrint then, maybe there is some setting acting up in the machine settings?

Hoppaz, 2019-11-20:
I've updated octoprint but no success.

I've recognized that everything is fine without connected raspberry. If I connect it via usb cable and press "x home" on the display, the head will move about 5mm to the right. This repeats on every touch of the button. If I disconnect the usb cable again and press "x home", the head moves to the position expected.

davidramiro, 2019-12-08:
Sounds more like a serial/hardware issue to me. Or maybe just something on OctoPrint's side. Nothing in this firmware changes when connecting it to a USB host. Sounds silly, but are you using the USB cable provided by Anycubic? That one is known to be of bad quality and can cause all sorts of issues.

Other than that, I will probably not be able to help you any further as I can't see how connecting an USB lead is causing the display to malfunction.

I'll close this for now, but please report back if you are able to narrow down the issue!

closed 2019-12-08 by davidramiro