davidramiro/Marlin-Ai3M #61: [BUG] 12864 LCD // RepRap // Rotary Encoder does not work

danizwam, 2019-12-02:
Hi,

i have an problem with this Firmware and my Rotary Encoder on my 12864 LCD.

I want to upgrade my Anycubic i3 Mega with a Piezo Orion and therefore i want to update the Firmware.

Currently i am on https://github.com/derhopp/Marlin-with-Anycubic-i3-Mega-TFT and with this version everything works fine.

I am using this LCD https://www.amazon.de/gp/product/B01LO20XGS/ and uncommented in the configuration.h this line "#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER".

The LCD works and i can click the rotary encoder to enter the first submenu (Info Screen, Prepare, Control..) but i cannot chose between them. I rotate the knob, but nothing happens.

For testing purposes I just downloaded the firmware of DerHopp and changed only this one line "#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" (and adjusted the AnycubicTFT.h (it has a problem with 'ftostr32 out of scope' im REPRAP_XXX is activated) and it all works fine.

I am referring to DerHopp, because he is mentioned on the first page.

Am i missing something?

Greetings, Daniel

davidramiro, 2019-12-02:
Did you use the master branch? If so, please try the one specifically configured for 12864: 

https://github.com/davidramiro/Marlin-Ai3M/tree/12864-full

danizwam, 2019-12-02:
Hi,

i just downloaded the mentioned 12864-Version, checked if the 
    #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 
is active and flashed it. 

I still can't chose with the knob :/

davidramiro, 2019-12-02:
Try downloading this zip and overwrite your pins_TRIGORILLA_14.h with this one.

[pins.zip](https://github.com/davidramiro/Marlin-Ai3M/files/3913528/pins.zip)

danizwam, 2019-12-02:
Works like charm ;)

Thank you very much for your quick help!

Now i can make my changes to support the Piezo Orion ;)

davidramiro, 2019-12-02:
Glad to help!

As for the piezo probe, you might want to check out this fork of my firmware: https://github.com/kulfuerst/Marlin-Ai3M-Piezo

I recently was in contact with him, he modded my firmware to support a piezo probe. There were some weird glitches he had to overcome to get it to work, maybe his commit history can be of use for you too. Good luck!

closed 2019-12-02 by davidramiro

Meganium97, 2021-12-08:
> Did you use the master branch? If so, please try the one specifically configured for 12864:
> 
> https://github.com/davidramiro/Marlin-Ai3M/tree/12864-full

I've noticed that this branch hasn't been updated in a bit. Is it still safe to use?