davidramiro/Marlin-Ai3M #62: extruder not working

davi-12, 2019-12-06:
Hi Davide Ramiro, a problem has just happened to me, after a print lasting about 30 minutes, the extruder seems to no longer work ... I tried to use the function to insert the filament, and I tried to give a new print, the extruder toothed wheel is still, can you please help me?

davidramiro, 2019-12-06:
Send
```
M502
M500
```
to restore factory settings (back up your current settings with `M503`).
Then restart the printer and try again. Try printing via USB and watch the terminal, maybe there's an error message.

If it still doesn't work, switch to the official firmware. If it doesn't work there, it's a hardware issue.

davi-12, 2019-12-06:
Hi David, thanks for having replied, so first I make a backup of the settings with "m503", sorry but to make the backup I copy and paste the messages that are displayed? After the backup send "m502" and after "m500", in this way I go back to the factory settings? Or do I have to reinstall it after the "M502" and "M500" commands?

davidramiro, 2019-12-06:
Exactly, just copy the output from M503 and save it in a document. So you can paste it after. After M502+M500, you do not need to reinstall.

davi-12, 2019-12-06:
ok, when I press "filament in" on pronterface display it appears written "too long extrusion prevented"

davidramiro, 2019-12-06:
`too long extrusion prevented` often comes from a wrong slicer configuration regarding relative and absolute positioning, did you change any settings there recently?

davi-12, 2019-12-06:
I ran the M502 and M500 commands, now it works, but it seems very slow

davi-12, 2019-12-06:
maybe because it needs to be set again?

davi-12, 2019-12-06:
I use cura and I have not changed anything

davi-12, 2019-12-06:
Hi, I wanted to update the post, I set everything up manually again following the procedure, because I didn't feel safe to restore the backup, now everything would seem to work. when I send m503, "Filament settings: Disabled" is written on the second line, what does it mean? it's all ok?

davidramiro, 2019-12-07:
>I wanted to update the post, I set everything up manually again following the procedure, because I didn't feel safe to restore the backup, now everything would seem to work

Good to hear that. I can't really say why this would happen, but I suspected something in your configuration caused a glitch. Make sure to save those working settings somewhere. If you need to restore it, just send all the lines (one at a time) and finish with `M500`.

>when I send m503, "Filament settings: Disabled" is written on the second line, what does it mean? it's all ok?

That's fine.

Have a good one.

closed 2019-12-07 by davidramiro