davidramiro/Marlin-Ai3M #63: [BUG] SD Print Filament Change and Pause Freezes TFT, No resume working.

sharonna511, 2019-12-17:
**Describe the bug**
I tested now a lot, no pause method is working.
Tested:
SD Print > press pause onscreen, head goes park, tft button switches to continue, but is then not responding.
SD Print > M25 Pause in GCODE, head goes park, tft button switches to continue, but is then not responding.
SD Print > M600  in GCODE head goes park, filament gets unloaded, peeps a lot, tft button switches to continue, but is then not responding.

Any help to fix that appreciated, Thanks.

**To Reproduce**
See above

**Expected behavior**
On onscreen or M25 pause, resume should be working.
On M600 resume should be working.

**Printer (please complete the following information):**
 - Firmware version: # Anycubic i3 Mega / Mega-S Marlin 1.1.9 by davidramiro, self compiled
 - Printing method: SD
 - Model: i3 Mega

**M503 Output**
SENT: M503
READ: echo:  G21    ; (mm)
G21    ; (mm)
READ: 
READ: echo:Filament settings: Disabled
Filament settings: Disabled
READ: echo:  M200 D1.75
M200 D1.75
READ: echo:  M200 D0
M200 D0
READ: echo:Steps per unit:
Steps per unit:
READ: echo:  M92 X80.00 Y80.00 Z400.00 E92.60
M92 X80.00 Y80.00 Z400.00 E92.60
READ: echo:Maximum feedrates (units/s):
Maximum feedrates (units/s):
READ: echo:  M203 X500.00 Y500.00 Z6.00 E60.00
M203 X500.00 Y500.00 Z6.00 E60.00
READ: echo:Maximum Acceleration (units/s2):
Maximum Acceleration (units/s2):
READ: echo:  M201 X3000 Y2000 Z60 E10000
M201 X3000 Y2000 Z60 E10000
READ: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
READ: echo:  M204 P2500.00 R3000.00 T3000.00
M204 P2500.00 R3000.00 T3000.00
READ: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
READ: echo:  M205 Q20000 S0.00 T0.00 X5.00 Y5.00 Z0.40 E5.00
M205 Q20000 S0.00 T0.00 X5.00 Y5.00 Z0.40 E5.00
READ: echo:Home offset:
Home offset:
READ: echo:  M206 X0.00 Y0.00 Z0.00
M206 X0.00 Y0.00 Z0.00
READ: echo:Mesh Bed Leveling:
Mesh Bed Leveling:
READ: echo:  M420 S0 Z0.00
M420 S0 Z0.00
READ: echo:  G29 S3 X1 Y1 Z0.04000
G29 S3 X1 Y1 Z0.04000
READ: echo:  G29 S3 X2 Y1 Z0.06000
G29 S3 X2 Y1 Z0.06000
READ: echo:  G29 S3 X3 Y1 Z0.06000
G29 S3 X3 Y1 Z0.06000
READ: echo:  G29 S3 X4 Y1 Z0.02000
G29 S3 X4 Y1 Z0.02000
READ: echo:  G29 S3 X5 Y1 Z0.00000
G29 S3 X5 Y1 Z0.00000
READ: echo:  G29 S3 X1 Y2 Z0.14000
G29 S3 X1 Y2 Z0.14000
READ: echo:  G29 S3 X2 Y2 Z0.14000
G29 S3 X2 Y2 Z0.14000
READ: echo:  G29 S3 X3 Y2 Z0.10000
G29 S3 X3 Y2 Z0.10000
READ: echo:  G29 S3 X4 Y2 Z0.08000
G29 S3 X4 Y2 Z0.08000
READ: echo:  G29 S3 X5 Y2 Z0.08000
G29 S3 X5 Y2 Z0.08000
READ: echo:  G29 S3 X1 Y3 Z0.14000
G29 S3 X1 Y3 Z0.14000
READ: echo:  G29 S3 X2 Y3 Z0.16000
G29 S3 X2 Y3 Z0.16000
READ: echo:  G29 S3 X3 Y3 Z0.14000
G29 S3 X3 Y3 Z0.14000
READ: echo:  G29 S3 X4 Y3 Z0.12000
G29 S3 X4 Y3 Z0.12000
READ: echo:  G29 S3 X5 Y3 Z0.14000
G29 S3 X5 Y3 Z0.14000
READ: echo:  G29 S3 X1 Y4 Z0.12000
G29 S3 X1 Y4 Z0.12000
READ: echo:  G29 S3 X2 Y4 Z0.12000
G29 S3 X2 Y4 Z0.12000
READ: echo:  G29 S3 X3 Y4 Z0.12000
G29 S3 X3 Y4 Z0.12000
READ: echo:  G29 S3 X4 Y4 Z0.12000
G29 S3 X4 Y4 Z0.12000
READ: echo:  G29 S3 X5 Y4 Z0.14000
G29 S3 X5 Y4 Z0.14000
READ: echo:  G29 S3 X1 Y5 Z0.02000
G29 S3 X1 Y5 Z0.02000
READ: echo:  G29 S3 X2 Y5 Z0.04000
G29 S3 X2 Y5 Z0.04000
READ: echo:  G29 S3 X3 Y5 Z0.04000
G29 S3 X3 Y5 Z0.04000
READ: echo:  G29 S3 X4 Y5 Z0.04000
G29 S3 X4 Y5 Z0.04000
READ: echo:  G29 S3 X5 Y5 Z0.08000
G29 S3 X5 Y5 Z0.08000
READ: echo:Endstop adjustment:
Endstop adjustment:
READ: echo:  M666 Z0.00
M666 Z0.00
READ: echo:PID settings:
PID settings:
READ: echo:  M301 P15.94 I1.17 D54.19
M301 P15.94 I1.17 D54.19
READ: echo:  M304 P251.78 I49.57 D319.73
M304 P251.78 I49.57 D319.73
READ: echo:Linear Advance:
Linear Advance:
READ: echo:  M900 K0.00
M900 K0.00
READ: echo:Filament load/unload lengths:
Filament load/unload lengths:
READ: echo:  M603 L538.00 U555.00
M603 L538.00 U555.00
READ: ok

davidramiro, 2019-12-20:
Cannot reproduce any of those issues, anything special on your setup? Since you are using a self-compiled version, anything noteworthy that you changed? I'd recommend trying again with a precompiled release, at least that's what I am testing on.

sharonna511, 2019-12-20:
Yeah its strange, i figured you would have noticed such things, but it is so
Changed, just basic stuff, tmc2208 reversed plugs in printer, commented out the filament runout sensor, removed and add a few entries in special menu. But i must say that even before with the anycubic firmware when i put M25 pause in gcode resume did not work, tft just never displayed resume. Maybe i have a buggy or different tft version ?

davidramiro, 2019-12-20:
>before with the anycubic firmware when i put M25 pause in gcode resume did not work, tft just never displayed resume.

Yes, that is a feature I built into this version.

>commented out the filament runout sensor

Not sure if that might cause it, but try again with it enabled.

Also, you can uncomment `ANYCUBIC_TFT_DEBUG` in Configuration.h to get some more info via USB in the terminal.

sharonna511, 2019-12-20:
thanks i will test and report back.

davidramiro, 2020-01-10:
Could you find anything out?

sharonna511, 2020-01-10:
I did not have time over xmas/new year. I tested it now.
I had filament sensor and tft debug uncommented. Also i had the filament sensor connected and filament was in it.

m25 in gcode > parks, display says pause not continue, on press on pause not responding, then i press stop, the message came with ok and back arrow, pressed back arrow, but it stopped then, as if i pressed ok

So i don't know what is wrong there, would be nice it is was working, but it is not high priority for me, since i can always use my Kossel for filament swap things at the time.
But it is very strange...

log:
SENT: M105
READ: echo:enqueueing "M125"
enqueueing "M125"
READ: X:114.40 Y:95.60 Z:11.04 E:0.00 Count X:8225 Y:9130 Z:4351
READ: ok T:199.90 /200.00 B:60.02 /60.00 @:85 B@:38
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: ok T:201.16 /200.00 B:59.90 /60.00 @:67 B@:60
READ: ok T:201.16 /200.00 B:59.90 /60.00 @:67 B@:60
READ: ok T:201.16 /200.00 B:59.90 /60.00 @:67 B@:60
READ: ok T:201.16 /200.00 B:59.90 /60.00 @:67 B@:60
SENT: M105
READ: ok T:201.56 /200.00 B:59.96 /60.00 @:62 B@:49
SENT: M105
READ: ok T:201.74 /200.00 B:60.06 /60.00 @:60 B@:34
SENT: M105
READ: ok T:201.96 /200.00 B:60.12 /60.00 @:57 B@:23
SENT: M105
READ: ok T:201.83 /200.00 B:59.96 /60.00 @:59 B@:51
SENT: M105
READ: ok T:201.65 /200.00 B:59.99 /60.00 @:60 B@:47
SENT: M105
READ: ok T:201.12 /200.00 B:60.08 /60.00 @:66 B@:30
SENT: M105
READ: ok T:200.54 /200.00 B:60.07 /60.00 @:72 B@:30
SENT: M105
READ: ok T:200.31 /200.00 B:59.99 /60.00 @:72 B@:43
SENT: M105
READ: ok T:199.87 /200.00 B:60.05 /60.00 @:76 B@:32
SENT: M105
READ: ok T:199.77 /200.00 B:60.06 /60.00 @:75 B@:29
SENT: M105
READ: ok T:199.64 /200.00 B:59.96 /60.00 @:75 B@:46
SENT: M105
READ: ok T:199.67 /200.00 B:60.01 /60.00 @:74 B@:37
SENT: M105
READ: ok T:199.70 /200.00 B:60.05 /60.00 @:74 B@:30
SENT: M105
READ: ok T:199.80 /200.00 B:60.01 /60.00 @:73 B@:35
SENT: M105
READ: ok T:199.93 /200.00 B:60.10 /60.00 @:71 B@:17
SENT: M105
READ: ok T:200.00 /200.00 B:60.03 /60.00 @:71 B@:29
SENT: M105
READ: ok T:200.04 /200.00 B:60.06 /60.00 @:70 B@:23
SENT: M105
READ: ok T:200.13 /200.00 B:59.91 /60.00 @:69 B@:49
SENT: M105
READ: ok T:200.09 /200.00 B:60.02 /60.00 @:70 B@:30
SENT: M105
READ: ok T:199.97 /200.00 B:60.07 /60.00 @:72 B@:20
SENT: M105
READ: ok T:199.97 /200.00 B:59.92 /60.00 @:72 B@:45
SENT: M105
READ: ok T:199.97 /200.00 B:59.97 /60.00 @:72 B@:38
SENT: M105
READ: ok T:200.04 /200.00 B:60.03 /60.00 @:71 B@:28
SENT: M105
READ: ok T:199.77 /200.00 B:60.00 /60.00 @:75 B@:36
SENT: M105
READ: ok T:199.80 /200.00 B:60.06 /60.00 @:74 B@:25
READ: TFT Serial Command: A9
READ:  DEBUG: AI3M Pause State: 0
READ: DEBUG: Stopped and cleared
READ:  DEBUG: AI3M Pause State: 0
READ: TFT Serial Debug: SD print stopped... J16
READ:  DEBUG: AI3M Pause State: 0
READ: echo:enqueueing "G91"
enqueueing "G91"
READ: echo:enqueueing "G1 E-1 F1800"
enqueueing "G1 E-1 F1800"
READ: echo:enqueueing "G90"
enqueueing "G90"
SENT: M105
SENT: M105
SENT: M105
READ: DEBUG: SDSTOP: Park Z
READ: DEBUG: SDSTOP: Park XY
READ: echo:enqueueing "M84"
enqueueing "M84"
READ: echo:enqueueing "M27"
enqueueing "M27"
READ:  DEBUG: AI3M Pause State: 0
READ: ok T:201.29 /0.00 B:59.45 /0.00 @:0 B@:0
READ: ok T:201.29 /0.00 B:59.45 /0.00 @:0 B@:0
READ: ok T:201.29 /0.00 B:59.45 /0.00 @:0 B@:0
READ: Not SD printing
SENT: M105
READ: ok T:201.52 /0.00 B:59.28 /0.00 @:0 B@:0
SENT: M105
READ: ok T:200.49 /0.00 B:59.09 /0.00 @:0 B@:0
SENT: M105
READ: ok T:199.47 /0.00 B:59.01 /0.00 @:0 B@:0
SENT: M105
READ: ok T:198.22 /0.00 B:58.82 /0.00 @:0 B@:0
READ: TFT Serial Command: A23
SENT: M105
READ: ok T:196.58 /200.00 B:58.67 /50.00 @:47 B@:0
SENT: M105
READ: ok T:194.90 /200.00 B:58.58 /50.00 @:68 B@:0
SENT: M105
READ: ok T:193.39 /200.00 B:58.44 /50.00 @:87 B@:0
SENT: M105
READ: ok T:192.20 /200.00 B:58.27 /50.00 @:103 B@:0
SENT: M105
READ: ok T:191.81 /200.00 B:58.19 /50.00 @:110 B@:0

coolio986, 2020-01-20:
I also have this same issue described above. Tested with the same gcode.

andreymukha, 2020-04-05:
I and many people from the Russian chat of Anycubic I3 Mega have this problem as described above

JensW2000, 2020-04-20:
> Yes, that is a feature I built into this version.
>
>
> Not sure if that might cause it, but try again with it enabled.
>
> Also, you can uncomment `ANYCUBIC_TFT_DEBUG` in Configuration.h to get some more [...]

Here the same problem. Neither Continue nor Stop works when the printer is paused.
This affects both  - the "Pause button" and M600.

The error is gone when 
#define ANYCUBIC_FILAMENT_RUNOUT_SENSOR is declared again

DenizAnDes, 2021-10-31:
Hi! I fixed this bug in my Anycubic i3 mega.
1) modify function **void AnycubicTFTClass::StateHandler()** (file AnycubicTFT.cpp) :
 after **case ANYCUBIC_TFT_STATE_SDPAUSE_REQ:**
put this:
`case ANYCUBIC_TFT_STATE_SDPAUSE_REQ:
      if(ai3m_pause_state == 2)
      {
        ai3m_pause_state = 0;
        AnDesPause = true;
        #ifdef ANYCUBIC_TFT_DEBUG
          SERIAL_ECHOLNPGM("DEBUG ANDES: AnDesPause"); 
        #endif
        CommandScan();
        break;
      }`
2) modify function **void AnycubicTFTClass::GetCommandFromTFT()** (same file) 
`void AnycubicTFTClass::GetCommandFromTFT()
{
  char *starpos = NULL;
  while( (AnDesPause && !serial3_count) || (AnycubicSerial.available() > 0  && TFTbuflen < TFTBUFSIZE))
  {
    if(AnDesPause && !serial3_count)
    {
      serial3_char = '\n';
      serial3_count = 0;
      TFTcmdbuffer[TFTbufindw][serial3_count] = 'A';
      serial3_count++;
      TFTcmdbuffer[TFTbufindw][serial3_count] = '9';
      serial3_count++;
      TFTcmdbuffer[TFTbufindw][serial3_count] = '\r';
      serial3_count++;
      TFTcmdbuffer[TFTbufindw][serial3_count] = '\n';
      serial3_count++;
      AnDesPause = false;
    }
    else{
      serial3_char = AnycubicSerial.read();
    }`
3) in file AnycubicTFT.h
add private variable to class AnycubicTFTClass
`private:
  bool AnDesPause = false;`

good luck!