davidramiro/Marlin-Ai3M #64: Change speed of motherboard cooling fan

jancoow, 2019-12-21:
Hi,

Is it possible to reduce the speed of the motherboard cooling fan? I've changed it with a slightly bigger fan, so it's producing a lot more airflow then the original one. However, it's kinda loud at full speed (12v). Would it be possible to reduce the speed through the firmware itself?

davidramiro, 2019-12-21:
Sure thing, this is the line you'd want to change: https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/Configuration_adv.h#L203

jancoow, 2019-12-22:
Thanks, awesome! Is this only possible using the configuration file and compile it myself? Or can I change this on my current firmware as well?  (Using gcode?)

davidramiro, 2019-12-23:
You'd need to compile it yourself like this: https://github.com/davidramiro/Marlin-AI3M/wiki/Customization-&-Compiling

If you compile and flash the same version you've got on your printer right now, it will will keep your configuration.

leandroalbero, 2019-12-26:
Hello! Is extruder fan PWM capable? I've tried changing its speed on this line, but it leads to the code not compiling.
https://github.com/davidramiro/Marlin-Ai3M/blob/2f686a790499c8cfe943e8076ceadb4eb9ab5b97/Marlin/Configuration_adv.h#L247

```
Arduino: 1.8.10 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch/MarlinConfig.h:42:0,
                 from sketch/I2CPositionEncoder.cpp:31:
SanityCheck.h:1416:5: error: non-constant condition for static assertion
     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from sketch/fastio.h:50:0,
                 from sketch/HAL.h:33,
                 from sketch/MarlinConfig.h:39,
                 from sketch/I2CPositionEncoder.cpp:31:
fastio_1280.h:340:21: error: reinterpret_cast from integer to pointer
 #define DIO44_PWM   &OCR5CL
                     ^
sketch/fastio.h:91:31: note: in expansion of macro 'DIO44_PWM'
 #define _GET_TIMER(IO)        DIO ## IO ## _PWM
                               ^~~
sketch/fastio.h:103:31: note: in expansion of macro '_GET_TIMER'
 #define GET_TIMER(IO)         _GET_TIMER(IO)
                               ^~~~~~~~~~
sketch/SanityCheck.h:1416:19: note: in expansion of macro 'GET_TIMER'
     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~~~
In file included from sketch/MarlinConfig.h:42:0,
                 from sketch/AnycubicSerial.cpp:34:
SanityCheck.h:1416:5: error: non-constant condition for static assertion
     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from sketch/fastio.h:50:0,
                 from sketch/HAL.h:33,
                 from sketch/MarlinConfig.h:39,
                 from sketch/AnycubicSerial.cpp:34:
fastio_1280.h:340:21: error: reinterpret_cast from integer to pointer
 #define DIO44_PWM   &OCR5CL
                     ^
sketch/fastio.h:91:31: note: in expansion of macro 'DIO44_PWM'
 #define _GET_TIMER(IO)        DIO ## IO ## _PWM
                               ^~~
sketch/fastio.h:103:31: note: in expansion of macro '_GET_TIMER'
 #define GET_TIMER(IO)         _GET_TIMER(IO)
                               ^~~~~~~~~~
sketch/SanityCheck.h:1416:19: note: in expansion of macro 'GET_TIMER'
     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~~~
In file included from sketch/MarlinConfig.h:42:0,
                 from sketch/G26_Mesh_Validation_Tool.cpp:27:
SanityCheck.h:1416:5: error: non-constant condition for static assertion
     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from sketch/fastio.h:50:0,
                 from sketch/HAL.h:33,
                 from sketch/MarlinConfig.h:39,
                 from sketch/G26_Mesh_Validation_Tool.cpp:27:
fastio_1280.h:340:21: error: reinterpret_cast from integer to pointer
 #define DIO44_PWM   &OCR5CL
                     ^
sketch/fastio.h:91:31: note: in expansion of macro 'DIO44_PWM'
 #define _GET_TIMER(IO)        DIO ## IO ## _PWM
                               ^~~
sketch/fastio.h:103:31: note: in expansion of macro '_GET_TIMER'
 #define GET_TIMER(IO)         _GET_TIMER(IO)
                               ^~~~~~~~~~
sketch/SanityCheck.h:1416:19: note: in expansion of macro 'GET_TIMER'
     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~~~
In file included from sketch/MarlinConfig.h:42:0,
                 from sketch/AnycubicTFT.cpp:27:
SanityCheck.h:1416:5: error: non-constant condition for static assertion
     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from sketch/fastio.h:50:0,
                 from sketch/HAL.h:33,
                 from sketch/MarlinConfig.h:39,
                 from sketch/AnycubicTFT.cpp:27:
fastio_1280.h:340:21: error: reinterpret_cast from integer to pointer
 #define DIO44_PWM   &OCR5CL
                     ^
sketch/fastio.h:91:31: note: in expansion of macro 'DIO44_PWM'
 #define _GET_TIMER(IO)        DIO ## IO ## _PWM
                               ^~~
sketch/fastio.h:103:31: note: in expansion of macro '_GET_TIMER'
 #define GET_TIMER(IO)         _GET_TIMER(IO)
                               ^~~~~~~~~~
sketch/SanityCheck.h:1416:19: note: in expansion of macro 'GET_TIMER'
     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~~~
In file included from sketch/MarlinConfig.h:42:0,
                 from sketch/M100_Free_Mem_Chk.cpp:49:
SanityCheck.h:1416:5: error: non-constant condition for static assertion
     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from sketch/fastio.h:50:0,
                 from sketch/HAL.h:33,
                 from sketch/MarlinConfig.h:39,
                 from sketch/M100_Free_Mem_Chk.cpp:49:
fastio_1280.h:340:21: error: reinterpret_cast from integer to pointer
 #define DIO44_PWM   &OCR5CL
                     ^
sketch/fastio.h:91:31: note: in expansion of macro 'DIO44_PWM'
 #define _GET_TIMER(IO)        DIO ## IO ## _PWM
                               ^~~
sketch/fastio.h:103:31: note: in expansion of macro '_GET_TIMER'
 #define GET_TIMER(IO)         _GET_TIMER(IO)
                               ^~~~~~~~~~
sketch/SanityCheck.h:1416:19: note: in expansion of macro 'GET_TIMER'
     static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~~~
exit status 1
non-constant condition for static assertion

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

davidramiro, 2020-01-10:
Since SanityCheck is throwing errors, I would reckon you can't use that fan with PWM.

closed 2020-01-10 by davidramiro