davidramiro/Marlin-Ai3M #65: [BUG] Mesh Bed Leveling not working as expected

fenexomega, 2019-12-24:
**Describe the bug**
Mesh bed leveling is not working on a brand new i3 Mega S. It worked the first time I set after flashing the custom Marlin, but after calibrating the z screws and redoing the mesh bed leveling again it seems to not be using the settings, even after putting the custom G-Code on cura.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Printer
2. Go to Print
3. Special Menu > Start Mesh Bed Leveling
4. Do the entire process, let the printer beep, wait 20 seconds and save to EEPROM.
5. Reset, Shutdown e Turn on the printer again.
6. Print anything

**Expected behavior**
A nice printed benchy (got it one time).

**Printer (please complete the following information):**
 - Firmware version: [v1.4.6 stable]
 - Printing method CURA/SD
 - Model Mega-S

**M503 Output**
```
echo:  G21    ; (mm)
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E412.90
echo:Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z6.00 E30.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y2000 Z60 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R1500.00 T3000.00
echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Mesh Bed Leveling:
echo:  M420 S0 Z0.00
echo:  G29 S3 X1 Y1 Z0.10000
echo:  G29 S3 X2 Y1 Z-0.06000
echo:  G29 S3 X3 Y1 Z-0.10000
echo:  G29 S3 X4 Y1 Z-0.06000
echo:  G29 S3 X5 Y1 Z0.14000
echo:  G29 S3 X1 Y2 Z0.08000
echo:  G29 S3 X2 Y2 Z-0.04000
echo:  G29 S3 X3 Y2 Z-0.14000
echo:  G29 S3 X4 Y2 Z-0.10000
echo:  G29 S3 X5 Y2 Z0.12000
echo:  G29 S3 X1 Y3 Z0.06000
echo:  G29 S3 X2 Y3 Z-0.06000
echo:  G29 S3 X3 Y3 Z-0.16000
echo:  G29 S3 X4 Y3 Z-0.10000
echo:  G29 S3 X5 Y3 Z0.10000
echo:  G29 S3 X1 Y4 Z0.06000
echo:  G29 S3 X2 Y4 Z-0.08000
echo:  G29 S3 X3 Y4 Z-0.16000
echo:  G29 S3 X4 Y4 Z-0.12000
echo:  G29 S3 X5 Y4 Z0.08000
echo:  G29 S3 X1 Y5 Z0.06000
echo:  G29 S3 X2 Y5 Z-0.06000
echo:  G29 S3 X3 Y5 Z-0.14000
echo:  G29 S3 X4 Y5 Z-0.12000
echo:  G29 S3 X5 Y5 Z0.10000
echo:Endstop adjustment:
echo:  M666 Z0.00
echo:PID settings:
echo:  M301 P15.94 I1.17 D54.19
echo:  M304 P251.78 I49.57 D319.73
echo:Linear Advance:
echo:  M900 K0.00
echo:Filament load/unload lengths:
echo:  M603 L538.00 U555.00
```

davidramiro, 2019-12-24:
Since you adjusted the Z-screws and it stopped working after, I'd try having a look at those again. Did you lower or raise them? Maybe you are lacking space now for the Mesh bed leveling to compensate.

fenexomega, 2019-12-24:
As far as I remember I lowered one of them to leave both at the same height. They seem OK now as far as I can measure.

davidramiro, 2019-12-24:
I see... Should not be an issue in theory. But maybe try raising them both by the same amount and try leveling again.

fenexomega, 2019-12-24:
Crazy right? I mean, as far as the machine knows, 0 Z is where I set the screws on. It shouldn't matter if is 5mm high or low for the "default" or expected value. It should compensate anyways, as the MMBL is set using position relatives to Z and I'm the one who set theses values. I always felt the resistance on the paper while doing the MMBL and even left it well thigh to see if it the print would be coming understruded, then it doesn't. It comes with like it ignored the MMBL. And I even opened the gcode to check if the custom gcode was there.

I flashed this firmware because my X-rods bent or my bed is warped (don't know which) on a brand new Mega S.  I got so hyped when the benchy printed (although with some imperfections)

fenexomega, 2019-12-24:
Later I will try to flash de firmware again, use my calibration settings and do the MMBL to see if changes anything.

davidramiro, 2019-12-24:
Crazy indeed. I too adjusted my Z endstop screws since they were misaligned and MMBL continued working, I don't see why it wouldn't either.

It's just the timing on your end that is interesting, since it worked once and stopped as soon as you adjusted them.

Yeah, flashing again might be a good idea. Reset the EEPROM with `M502` and `M500` after flashing to be sure.

davidramiro, 2019-12-24:
By the way, how are you experiencing the Mesh Leveling not to work? Is it just printing like there is no Mesh at all, i.e. just like on the manufacturers firmware or is the nozzle generally too high but still compensating? For troubleshooting, you could slice two bed level tests, one with `M420 S1` and one without.

fenexomega, 2019-12-24:
Is actually worst than the manufactures firmware. The center is too high, even if I making sure than is too thigh to what I consider to be ideal when doing MMBL.

I will try to use the tests you recommended. Do you recommend any bed level test obj?

davidramiro, 2019-12-24:
Interesting. Something like [this](https://www.thingiverse.com/thing:2797420) should work well.

fenexomega, 2019-12-28:
Hello David. I finally got rid of the problem. And it wasn't on the printer side (aside the warped bed) or firmware, but from the **slicer**. Cura wasn't properly configured. 

The problem was that I had set the wrong gcode flavor. It was **marlin**, then I had set the **reprap** instead (as shown in the manual). The manual confused me as it seems to instruct to a old version of Cura, where RepRap is shown as "RepRap (marlin/sprinter)". *facepalm*

I just did the bed MMBL again is it working great. Just printing my Diablo Mini as I type this. 

Thank you for working on this firmware and for letting me have a chance to use the entire potential of my machine, working around the manufacturer's mistake.

closed 2019-12-28 by fenexomega

davidramiro, 2019-12-28:
Thanks for letting me know! Yes, it is always some small stuff like this that causes the weirdest errors. 🙂 
 Glad it's working out now.