davidramiro/Marlin-Ai3M #69: Attempting to compile in Arduino- Multiple instances of xyz "not declared in this scope" [BUG]

Totalgeek9224, 2020-01-10:
**Describe the bug**
When attempting to compile the TMC2208 version of the Marlin using ArduinoIDE, multiple instances of a variable no declared occurs. 
The first was"wait_for_user" was not declared in this scope" and was fixed by adding #include "emergency_parser.h" to AnycubicTFT.cpp
I can't remember what the next issue was , but it had something to do with "reset_heater_idle_timer" not being a part of "class Temperature"
The after amending that , now I'm receiving this error upon compile "'heater_idle_timeout_ms' was not declared in this scope"

**Photos of the print / output of the terminal**
If applicable, add photos or console outputs to help explain your problem.
[image]

**Printer (please complete the following information):**
 - Firmware version: Release 1.4.6 
 - Printing method N/A
 - Model Anycubic i3 Mega

**M503 Output**
Please paste the complete output of the `M503` command here.

davidramiro, 2020-01-10:
Are you trying to compile the latest files of the `master` branch? Also, did you select the `ATmega2560` under `Tools` -> `Processor` and `Board`?

If that is all the case, anything specific you changed within the code? It should not need any changes to compile cleanly.

Totalgeek9224, 2020-01-10:
Yeah. To my knowledge, it's the lastest branch, and its definitely trying to compile to ATmega2560
[image]

I've compiled with this setup before, so I don't think the fault is there. 

I HAVE changed some things in the code, mainly to keep the functionality i had already (UBL with BLtouch, but this mainly consisted of altering some values here and there, and adding a couple of lines of code to ensure everything from my previous config was there. Let me try downloading a clean version and try compiling that

Totalgeek9224, 2020-01-10:
Fresh copy compiled and uploaded. No touchscreen functionality though. 
Seems like some of my changes must've busted the previous attempts. Any suggestions on how to avoid the same mistake? 
thanks!

davidramiro, 2020-01-10:
Well, if you want to be thorough, you could fork this repo, set it up locally and then track every single change via commits.

Otherwise, you could just use a diff checking tool to compare the 'clean' AnycubicTFT.cpp and the one with your changes. Maybe it's something obvious like an accidentally deleted line or something.

Totalgeek9224, 2020-01-10:
I appreciate your timely support. I'll try the forking method, and if i find out where i went wrong, i'll drop you a message, maybe could help somewhere in the future :)

davidramiro, 2020-01-10:
No problem! I'll close here for now since it's not a bug of the firmware per se, but feel free to report back on here.

closed 2020-01-10 by davidramiro