davidramiro/Marlin-Ai3M #70: [BUG] Wait for Bed / Extruder to reach temp does not work

Jensemann, 2020-01-16:
**Issue****
I'm having issues in using the following GCODE sequence to heat up bed & extr on startup, using octoprint:
----------------------------- GCODE -----------------------------
M106 S0 ;fan off
G28 XY ;home
M104 S[extruder0_temperature] T0 ; set temp for extruder 0 do not wait
M140 S[bed0_temperature] ; set temp for bed do not wait
M190 S[bed0_temperature] ; wait for bed to reach temp
M109 S[extruder0_temperature] T0 ; wait for extruder 0 to reach temp
G28 ;home all axis

G91; Set all axes to relative 
G1 X2.0 Y2.0 F7200 ; Fahre in Ausgangsposition
G1 Z0.1 ; Nozzle senken um 0.1mm
G1 X2.0 F7200 ; Material abstreifen
G1 Z0.2 ; Hebe Nozzle an auf 0.2mm
G1 Y150 E20 F600 ; Linie ziehen
G90
---------------------------------------

*** firmware ****

FW: Recv: FIRMWARE_NAME:Marlin 1.1.9 (Github, davidramiro) MARLIN-AI3M_VERSION:v1.4.6 SOURCE_CODE_URL:https://github.com/davidramiro/Marlin-AI3M PROTOCOL_VERSION:1.0 MACHINE_TYPE:Anycubic i3 Mega EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

*** printer ***
Anycubic i3Mega S. in use with Octopi.

**Expected behavior**
printer should wait till heated.

What happend is that printer starts to draw the line (cold extrusion prevented):
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M106 S0*102
Recv: ok
Send: N2 G28 XY*48
Recv:  T:120.54 /0.00 B:48.08 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:119.05 /0.00 B:47.94 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:117.64 /0.00 B:47.81 /0.00 @:0 B@:0
Recv: X:-5.00 Y:0.00 Z:20.30 E:-3.00 Count X:-400 Y:0 Z:8121
Recv: ok
Send: N3 M104 S[extruder0_temperature] T0*10
Recv: ok
Send: N4 M140 S[bed0_temperature]*55
Recv: ok
Send: N5 M190 S[bed0_temperature]*59
Recv: ok
Send: N6 M109 S[extruder0_temperature] T0*2
Recv: ok
Send: N7 G28*20
Recv:  T:116.11 /0.00 B:47.75 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:114.62 /0.00 B:47.70 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:113.19 /0.00 B:47.55 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:111.73 /0.00 B:47.43 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: X:-5.00 Y:0.00 Z:0.00 E:-3.00 Count X:-400 Y:0 Z:0
Recv: ok
Send: N8 G91*25
Recv: ok
Send: N9 G1 X2.0 Y2.0 F7200*67
Recv: ok
Send: N10 G1 Z0.1*76
Recv: ok
Send: N11 G1 X2.0 F7200*47
Recv: ok
Send: N12 G1 Z0.2*77
Recv: ok
Send: N13 G1 Y150 E20 F600*96
Recv: echo: cold extrusion prevented
Recv: echo: cold extrusion prevented
Recv: ok

**M503 Output**
Send: M503
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E421.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E30.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X2000 Y1500 Z60 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P4000.00 R3000.00 T4000.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 Q20000 S0.00 T0.00 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:  G29 S3 X1 Y1 Z0.04000
Recv: echo:  G29 S3 X2 Y1 Z0.06000
Recv: echo:  G29 S3 X3 Y1 Z0.06000
Recv: echo:  G29 S3 X4 Y1 Z0.04000
Recv: echo:  G29 S3 X5 Y1 Z0.00000
Recv: echo:  G29 S3 X1 Y2 Z0.00000
Recv: echo:  G29 S3 X2 Y2 Z0.02000
Recv: echo:  G29 S3 X3 Y2 Z0.02000
Recv: echo:  G29 S3 X4 Y2 Z0.04000
Recv: echo:  G29 S3 X5 Y2 Z0.02000
Recv: echo:  G29 S3 X1 Y3 Z0.00000
Recv: echo:  G29 S3 X2 Y3 Z0.02000
Recv: echo:  G29 S3 X3 Y3 Z0.02000
Recv: echo:  G29 S3 X4 Y3 Z0.04000
Recv: echo:  G29 S3 X5 Y3 Z0.04000
Recv: echo:  G29 S3 X1 Y4 Z-0.02000
Recv: echo:  G29 S3 X2 Y4 Z0.00000
Recv: echo:  G29 S3 X3 Y4 Z0.02000
Recv: echo:  G29 S3 X4 Y4 Z0.02000
Recv: echo:  G29 S3 X5 Y4 Z0.02000
Recv: echo:  G29 S3 X1 Y5 Z0.02000
Recv: echo:  G29 S3 X2 Y5 Z0.06000
Recv: echo:  G29 S3 X3 Y5 Z0.04000
Recv: echo:  G29 S3 X4 Y5 Z0.02000
Recv: echo:  G29 S3 X5 Y5 Z0.02000
Recv: echo:Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P15.94 I1.17 D54.19
Recv: echo:  M304 P251.78 I49.57 D319.73
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L538.00 U555.00
Recv: ok

davidramiro, 2020-01-16:
You are not submitting any valid temperature command.

`[bed0_temperature]` and `[extruder0_temperature]` are placeholders from a particular slicer. Did you by any chance copy the start script from somewhere?

From the transmitted GCode: `M109 S[extruder0_temperature] T02` is not a valid syntax for example. Replace that with `M109 S200` (or whatever temp you want to print on) in the GCode file and it will work.

This is not a firmware bug but rather an incorrectly configured start script.

davidramiro, 2020-02-04:
Since this doesn't seem like a bug per se, I'm closing this for now. Please report back if you continue to run into issues.

closed 2020-02-04 by davidramiro