davidramiro/Marlin-Ai3M #72: Manual mesh bed levelling dont work from lcd panel

Maxeemo, 2020-01-28:
When I enter the print menu to choose special menu the printer freeze

tlow92, 2020-01-28:
You probably forgot to delete the chinese named folder that is on the stock sd card.
Chinese characters break the special menu.

davidramiro, 2020-02-04:
Correct. This is also mentioned in the description:

Special characters on any file or folders name on the SD card will cause the file menu to freeze. Simply replace or remove every special character (Chinese, Arabic, Russian, accents, German & Scandinavian umlauts, ...) from the name. Symbols like dashes or underscores are no problem. Important note: On the SD card that comes with the printer there is a folder with Chinese characters in it by default. Please rename or remove it.

closed 2020-02-04 by davidramiro