davidramiro/Marlin-Ai3M #73: [FR] Activate M73 command (status percentage)

trunneml, 2020-02-14:
**Describe your suggestion.**
Marlin has the M73 command. And the i3 Mega has the status menu (Setup->Status) where it shows the status when printing from SD. Is it possible to activate the M73 command to show the percentage in the status menu.

**Reasoning**
 Would be useful when printing via octoprint or over usb.

**References, context or examples.**
https://marlinfw.org/docs/gcode/M073.html

davidramiro, 2020-02-16:
That's a great idea. Sadly there is one issue preventing me from implementing this: As far as I can see it, there is no way to make the screen show the print status window without actually starting an SD print. Also the bottom right text area can only be updated with very specific messages ("SD card inserted", "Filament runout", etc.). 

So right now, I don't see any way of including a status percentage. But the idea is good and it would be easy to show on the regular print screen, so I will check these days if I can force it to show.

Khaos66, 2022-02-15:
I was going to create the very same issue.
It's sad, that it doesn't seem possible...

Also [M117](https://marlinfw.org/docs/gcode/M117.html) would have been usefull. But will not be possible for the same reason I guess